arduino/Arduino #5922: Bad data in library_index.json

njh, 2017-01-28:
Hello,

I have just experienced a problem with library_index.json that powers the Arduino Library Manager, that I also use to generate http://www.arduinolibraries.info/:

```json
    {
      "name": "",
      "version": "1.2.1",
      "author": "NicoHood",
      "maintainer": "NicoHood \u003cblog@NicoHood.de\u003e",
      "sentence": "Connect all your favourite Nintendo Controllers to your Arduino.",
      "paragraph": "Currently supported: Gamecube, N64. Carefully read the wiki pages to know how to setup the controller connection.",
      "website": "https://github.com/NicoHood/Nintendo",
      "category": "Signal Input/Output",
      "architectures": [
        "avr"
      ],
      "types": [
        "Contributed"
      ],
      "url": "http://downloads.arduino.cc/libraries/NicoHood/-1.2.1.zip",
      "archiveFileName": "-1.2.1.zip",
      "size": 234305,
      "checksum": "SHA-256:23a2c327ab9890bf941e461382a8a4ff78df64b453e9b36733046bf1ce3970fa"
    },
```

The library doesn't have a name and the download URL is missing a name too. There isn't anything obviously wrong with ```library.properties``` from @NicoHood :

https://github.com/NicoHood/Nintendo/blob/1.2.1/library.properties

Thanks,

nick.

NicoHood, 2017-01-28:
Wow and the email is also parsed wrong. Now i know why i get spam emails to the u003cblog email address. Interesting how bots search for emails.

njh, 2017-01-28:
* \u003c is Unicode for <
* \u003e is Unicode for >

So it is perfect valid JSON, but not sure why the JSON generator did it - maybe so it can be inserted into HTML/XML without confusing parsers.

njh, 2017-01-28:
Hm, seems to be fixed now. I wonder what changed.

```json
    {
      "name": "Nintendo",
      "version": "1.2.1",
      "author": "NicoHood",
      "maintainer": "NicoHood \u003cblog@NicoHood.de\u003e",
      "sentence": "Connect all your favourite Nintendo Controllers to your Arduino.",
      "paragraph": "Currently supported: Gamecube, N64. Carefully read the wiki pages to know how to setup the controller connection.",
      "website": "https://github.com/NicoHood/Nintendo",
      "category": "Signal Input/Output",
      "architectures": [
        "avr"
      ],
      "types": [
        "Contributed"
      ],
      "url": "http://downloads.arduino.cc/libraries/NicoHood/Nintendo-1.2.1.zip",
      "archiveFileName": "Nintendo-1.2.1.zip",
      "size": 234945,
      "checksum": "SHA-256:c21733488643c231114143fdbdca560d9a3171eb3a0c6a035ddfff5e167be7e2"
    }
```

facchinm, 2017-01-30:
Unfortunately our jenkins configuration for the libraries crawler job only keeps the last ten artefacts so the offending one got lost over the weekend :sob: if it should happen again, please reopen the issue and we'll investigate further with the logs, thanks!

closed 2017-01-30 by facchinm

cmaglie, 2017-01-30:
It was the name field of the library that was accidentally edited to the empty string in our db. I've fixed it two days ago but forget to report here!