arduino/Arduino #5925: 2 repos for the same library

Rotzbua, 2017-01-28:
For example the Ethernet library 
https://github.com/arduino-libraries/Ethernet
https://github.com/arduino/Arduino/tree/master/libraries/Ethernet

So the question: where should be the PR/issues be?

njh, 2017-01-28:
See also #5908

Rotzbua, 2017-01-29:
@njh That's not the answer. I updated my description.

njh, 2017-01-29:
Sorry, I was not attempting to answer your question, I was simply trying to gather some of the issues together.

Both the Ethernet and Ethernet 2 shields have not been getting a lot of love recently - there are a lot of outstanding issues. While there are people willing to help solve problems and make improvements, there needs to be some leadership around what the plan is.

You issue highlights some fundamental problems around what the authoritative source of code is and where issues should be reported.

cmaglie, 2017-01-30:
Please send new PR/issues to https://github.com/arduino-libraries/Ethernet.

We have not removed yet the copy of the library in this repository to preserve existing pull requests, but we'll do soon, I'm in the process of evaluating various alternatives to move forward this library (including, but not all: #2325, @PaulStoffregen's [fork](https://github.com/PaulStoffregen/Ethernet), [Ethernet2](https://github.com/adafruit/Ethernet2) and a bunch of other patches people sent me).

closed 2017-01-30 by cmaglie

njh, 2017-01-30:
Good news, thanks @cmaglie 😃