arduino/Arduino #5928: Cancel button for verify/upload

johnwargo, 2017-01-30:
The IDE should display a cancel button (a small X button near the progress bar for example) that allows users to cancel verify/upload. I find myself accidentally clicking the upload button when I want to click verify (I know, I'm tired) and I have to wait for the complete build and a failed deploy before I can get back to work.

per1234, 2017-01-30:
The ability to cancel uploads was requested here:
https://github.com/arduino/Arduino/issues/3568
It was considered closed by adding the ability to restart the upload/compilation:
https://github.com/arduino/Arduino/issues/3771
but I don't consider that the same thing as an abort so I don't think #3568 should have been closed.

shiftleftplusone, 2017-01-30:
I don't see a "CANCEL" button in my IDE -  so how does it work?
https://github.com/arduino/Arduino/issues/3771 is not understandable - which actually is the solution?

facchinm, 2017-01-30:
The main problem tackled by #3771 is the fact that most programmers, if something goes wrong, lock the serial port indefinitely. 
Clicking the Upload button a second time, after that patch, kills the running programmer (and hopefully unlocks the port) and retries the upload.
Aborting an upload is only a matter of modifying the svg of the button and avoid relaunching the upload process. PRs on that topic are welcome and then the default behaviour can be discussed with some UX experts  :smile:

johnwargo, 2017-01-30:
Clicking the button again to cancel isn't intuitive, I never would have thought to do that and I've been a professional developer for more than 30 years. Switching the button image (which is what I think you're saying when you say change the svg, but that's not clear) would work, but then you'd have to change both buttons, and why do all that work when you can just add a Cancel button when Verify or Upload are in  progress.

Do you really think you need to speak with a UX expert to figure this out?  It's a simple problem and an simple, commonly implemented solution. I can list 10 reasons why it's needed:

1. I click the verify button then go, oh, wait, I need to change something else first.
2. I click the upload button only to realize that I don't have a board attached.
3. I click the upload button only to realize that the board is connected to a different port
4. I clicked either button accidently. 
5. I click either button, but then notice I have an error in my code.

I could go on and on and on. 

@shiftleftplusone there isn't a button, I'm suggesting that one be added (by someone other than me as I don't know the code well enough to add it myself).

facchinm, 2017-01-30:
The cancel button was suggested in the past and rejected with UX motivations. On the same topic, the Serial Plotter button near the Serial terminal one was rejected too with the same UX motivations. Create implements a modal cancel button for both compile and upload (it replaces the upload button only if this one was pressed) so I believe it's the way to go if we want to implement this feature

shiftleftplusone, 2017-01-30:
I already tried thet "click upload a 2nd time" on my Due without knowing that that that was a trick which actually was supposed to work, nonetheless, it didn't - when it once hung up, it didn't help to click it a 2nd time.
Most of the tomes I had to unplud the Die from USB, close the IDE, replug, restart, and re-upload.
And tbh,  that's a bit cumbersome.

matthijskooijman, 2017-01-31:
I seem to recall that the upload/compilation process isn't actually properly tracked in the code right now. I think when you click the upload or verify again, it will just run two processes concurrently, which will almost certainly not behave the way you want to. I think the code that runs the verify/upload processes would benefit from a big cleanup/refactor (there is a lot of legacy code still from processing there that serves no purpose anymore), keeping in mind the need to more properly abort a running process.

A related issue with aborting avrdude is #3569.

Blueshawk, 2017-11-20:
Aborting by restarting isn't aborting, it's restarting! :P 
  Currently, a bigger and more time consuming bug related to this results in the port timeout retry running amok even if you restart the upload process. The restart is failing to end the previous upload call or close the port and so hitting the upload button again before waiting till the reties all fail just piles on to the error. 
 This is a serious problem if you have added an RC reset timer to prevent the automatic reset on the Arduino as we must now wait and wait for all the timeouts to finish. No amount of disco or button yoga has made my system sync once this loop gets started.(unrelated hint:add a post compile sync pause button or at least an indicator) I'm currently removing the shield and thus the reset cap for every program change, about every 5 minutes during motor troubleshooting.

TDuxis, 2022-01-19:
Yes, please fix this.

Blueshawk, 2022-01-20:
At this point I think we'd all be happy if they just stopped trying to batter an open and locked port 9 more times after a failed sync. Just stop if sync fails and close the port. That would save loads of time and let us retry more quickly.

per1234, 2022-12-03:
Now tracked at https://github.com/arduino/arduino-ide/issues/1199

closed 2022-12-03 by per1234