arduino/Arduino #5931: WireWrapper library

SMFSW, 2017-01-31:
WireWrapper library (derived from cI2C library):
https://github.com/SMFSW/WireWrapper

shiftleftplusone, 2017-01-31:
I filed  an issue about
misleading: 400 kHz is Fast speed, not Low Speed !

could you additionally please work out a little more about the pros and cons to Wire.h ?

SMFSW, 2017-01-31:
Hello,

thanks for the feedback, I'll update speed in the day (for both WireWrapper & cI2C libs).

I'll try to explain some more about Pros & Cons to Wire.h on both WireWrapper & cI2C libraries.
About WireWrapper, it basically automates calls to Wire lib functions to perform read/write transactions in one call.

shiftleftplusone, 2017-01-31:
now I see, thank you very much for considering the changes!
kind regards!

SMFSW, 2017-02-01:
changes done!

cmaglie, 2017-02-03:
Library added, thanks

closed 2017-02-03 by cmaglie