arduino/Arduino #5932: Incomplete install script

chrisspen, 2017-01-31:
The `install.sh` Linux script included in the latest release doesn't appear to actually install anything, other than some unnecessary desktop icons. Ideally, this should install the Arduino IDE itself into the proper location.

Assuming you extract the tarball into `~/arduino-1.8.1`, installing should at minimum do:

    sudo mv ~/arduino-1.8.1 /usr/share/arduino
    sudo ln -s /usr/share/arduino/arduino /usr/local/bin/

This is the location the current Ubuntu Arduino package uses (although it's Arduino IDE version is ancient), and allows all users easy access to the IDE without worrying about setting up their PATH or other user session variables.

per1234, 2017-02-01:
Possible duplicate of https://github.com/arduino/Arduino/issues/5366
Also see: https://github.com/arduino/Arduino/issues/5780

tttapa, 2017-02-02:
This is a script I wrote to install the Arduino IDE on Ubuntu:
```bash
# Arduino installer

clear
installdir=/usr/share/
read -p "Install Arduino IDE? [Y/n]: " inst
if [ "$inst" = y ] || [ "$inst" = Y ]
then 
    read -p "Please enter a version number (e.g. 1.8.1): " version
    cd /tmp/
    if [ ! -e "arduino-$version-linux64.tar.xz" ]
    then
        echo "Downloading Arduino $version ..."
        wget "https://downloads.arduino.cc/arduino-$version-linux64.tar.xz"
    fi
    echo "Extracting ..."
    tar -xf "arduino-$version-linux64.tar.xz"
    sudo mv "arduino-$version/" "$installdir/arduino-$version"
    cd "$installdir/arduino-$version"
    ./install.sh
    if [ -e "/usr/bin/arduino" ]
    then
        rm "/usr/bin/arduino"
    fi
    sudo ln -s "arduino" "/usr/bin/arduino"
    sudo usermod -a -G dialout $USER
fi
```
Hope this helps!
Pieter

chrisspen, 2017-02-03:
@tttapa Thanks, that's really helpful.

facchinm, 2017-02-07:
@tttapa , I'd also add an `apt-get uninstall modemmanager` since it conflicts with most unofficial boards (and it causes a lot of headaches to Linux users)

tttapa, 2017-02-07:
@facchinm , Didn't know that (only ever used official boards or boards with genuine FTDI chips).
I edited the script to disable ModemManager (and added sudo to remove existing `/usr/bin/arduino` symlink).