arduino/Arduino #5934: Allow specifying custom library paths via the command line

chrisspen, 2017-01-31:
One of the remaining holes in the command line interface is the inability to specify custom library paths.

Many people like Arduino, but prefer to compile through the command line and find the IDE inconvenient and a roadblock to properly managing a project in git. For example, if you've committed libraries to your git project, the IDE can't find these until you install them to your global sketchbook...but now they're no longer under source control. Even worse, if you have multiple Arduino projects, and they use different versions of a library, you now have a version conflict.

[Several people](http://arduino.stackexchange.com/questions/8651/loading-local-libraries) have found [various hacks](http://stackoverflow.com/questions/4705790/keeping-all-libraries-in-the-arduino-sketch-directory), using symlinks or makefiles to [workaround](http://stackoverflow.com/questions/8189306/command-line-arduino-compiling-and-downloading) this problem, but adding an option via the command line to simply allow users to specify a local library path would greatly simplify this.

Tools like [ino](http://inotool.org/) and [ano](https://github.com/scottdarch/Arturo) have implemented this more directly by hacking Arduino's build process to explicitly look in a local `./lib` folder, but these projects are now unmaintained and no longer support the current Arduino version. These tools were developed back when Arduino didn't have an official command line. Now that it does, it makes more sense to expand the official command line interface rather than maintain a separate fragile tool that'll break with every new Arduino release.

facchinm, 2017-02-01:
Hi Chris, can you expand on that? That's true, the CLI for Java IDE doesn't have this ability, but `arduino-builder` does, and using it as a standalone CLI tool gives you a lot more control over the build variables.
If you need to upload the compiled code it's not that easy, but in the long run I would prefer another standalone uploader which you can pipe with the builder instead than adding another flag to the java IDE CLI.

chrisspen, 2017-02-02:
I've never heard of arduino-builder. Is there any documentation or example usage for it? The list of options on its [Github page](https://github.com/arduino/arduino-builder) don't mention how you specify the actual source files to compile. Does it accept arguments? Googling "arduino-builder" brings up mostly results about some arduino GUI tool.

facchinm, 2017-02-02:
[This](https://github.com/arduino/arduino-builder/wiki/Doing-continuous-integration-with-arduino-builder) wiki page shows a compilation example; the target sketch is the only argument without any flag.
Another way to extract an example commandline is by running the IDE GUI with verbose compilation enabled. This way, one of the first line will show the commandline used by the IDE, which you can tune as you need.

chrisspen, 2017-02-02:
Thanks. That's exactly what I was looking for.

closed 2017-02-07 by facchinm

cjheath, 2019-04-15:
This does not work for me. I have more than one client, and each has proprietary libraries for their own projects. I cannot reasonably keep these under their source code control *and* copy them to my sketches folder.

Please add the ability to specify one or more project libraries directory.