arduino/Arduino #5935: Please include my DCF77 library to the library manager

udoklein, 2017-01-31:
Hi everyone,

according to the FAQ https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ I need to create an issue for my library to be picked up by the library manager.

The most up to date version of my library can be found here:

https://github.com/udoklein/dcf77/releases/

The current version 3.1.4 is the first version to include a compliant properties file. The library is a high quality dcf77 decoder. It enjoys some popularity by now. Would you please include my contribution to the library manager?

Thanks,
Udo

cmaglie, 2017-02-03:
Hi @udoklein
there is already a library called DCF77, so I can't add it your as is.

Really, this is the first time a name that a name clash happens... may I ask you to change the library name? I don't know, something like `Udoklein DCF77` or maybe something else that works for you.
Just changing the name in the library.properties should work (remember to fix the release tag too).

udoklein, 2017-02-04:
Hi @cmaglie this surprises me a little bit. The Other library is called DCF77 while mine is called dcf77. Since C is case sensitive I would have expected no clash. So this is a library manager thing. I do mind to rename it for the library manager though. Just one thing for clarification: is it really sufficient to change the library.properties.name ? I would really hate to change the directory and include names as this would break all current consumers of the library. Given the fact that C is case sensitive it definitely does not clash (at the C level) with the other library.

Can you confirm that changing the name will be sufficient? I would then change it to "dcf77 (blinkenlight.net)".

Another minor: would it be acceptable that I do not change the label this time? I actually wanted the first library manager compliant version to be an approximation of pi (3.1.4). So far nobody pulled this version thus I would like to move the label tag to the next commit.

Best regards, Udo

cmaglie, 2017-02-04:
> Since C is case sensitive I would have expected no clash. So this is a library manager thing.

It has nothing to do with C (as language) but with the case-sensitiveness of the filesystem, on Linux there is no problem, but Windows and Mac filesystems are case-insensitive, if you make a folder called "libraries/DCF77" you can access the same folder as "libraries/dcf77" or "libraries/DcF77".

http://stackoverflow.com/questions/7199039/file-paths-in-windows-environment-case-sensitive

> Can you confirm that changing the name will be sufficient? I would then change it to "dcf77 (blinkenlight.net)".

Yes you need only to change library.properties.

> Another minor: would it be acceptable that I do not change the label this time? I actually wanted the first library manager compliant version to be an approximation of pi (3.1.4). So far nobody pulled this version thus I would like to move the label tag to the next commit.

Yes, sure.

udoklein, 2017-02-04:
I changed the properties file as discussed and did NOT alter the semangic version. Once this release is included I will not mess anymore with the semantic versioning and stick to it as intended.

cmaglie, 2017-02-06:
@udoklein 
there is another issue, I've just discovered a problem with the parenthesis inside the name.
What happens is that the library manager will accept the name and install the library, but the IDE refuses to use the library with a popup like:

[image]

at the moment there are other two libraries that contains parenthesis inside the name, but nobody ever lamented this weird behavior, I'm wondering if the authors ever tried to install their own library.
This is something that should be fixed, and I'm reporting it in another issue.

In the meantime I have to ask you to change again the name and remove the parenthesis

`name=dcf77_(blinkenlight.net)`

to

`name=dcf77_blinkenlight.net`

Apologies for the back and forth, I understand that is really unlucky and annoying to hit two problems, the name duplicate and the parenthesis issue, in a row...

udoklein, 2017-02-07:
I renamed it to

name=dcf77_xtal

"xtal" because it is intended to be used with with crystal oscillators.  So I got rid of brackets and the dot to be one the safe side :)

Let's see if there are more surprises or if this does the trick.

cmaglie, 2017-02-07:
Done! thanks for your patience :-)

closed 2017-02-07 by cmaglie

udoklein, 2017-02-07:
Excellent. I installed both the other dcf77 and mine. Both compile properly even is the other one is installed. At least under Linux they do not clash. With windows I did not test though. If Windows is not case sensitive they might probably clash.

I think this OK.