arduino/Arduino #5940: Arduino doesn't save file with default name

antti202, 2017-02-02:
To reproduce:
1. Open Arduino
2. Save file with default name, "random" error comes up saying "The sketch already contains a file name..." and after clicking OK it appears to be saving but it really doesn't.
3. Close Arduino
4. Reopen Arduino...poof the file is gone.

Lost a days work because of this, saved 50x times thinking the file was saved properly.

per1234, 2017-02-02:
Which version of the Arduino IDE are you using? I suspect you're encountering https://github.com/arduino/Arduino/issues/5423, which was fixed 4 months and 3 releases ago.

Before submitting an issue report you should always:
- Test using the [hourly build of the Arduino IDE](http://www.arduino.cc/en/Main/Software#hourly) to be sure the bug hasn't already been fixed. 
- Do a search at: https://github.com/arduino/Arduino/issues?q= to make sure the bug hasn't already been reported.

antti202, 2017-02-02:
You are right, I am using 1.6.12. I thought that Arduino updates itself, I guess I was wrong.

closed 2017-02-02 by antti202

oqibidipo, 2017-02-02:
It would, if only somebody updated https://www.arduino.cc/latest.txt. It is stuck at 10601.