arduino/Arduino #5942: Supplied avrdude (v6.3) does not work with USBasp

mozgy, 2017-02-02:
Supplied avrdude ->

```
  -?                         Display this usage.

avrdude version 6.3, URL: <http://savannah.nongnu.org/projects/avrdude/>

C:\Program Files (x86)\Arduino\hardware\tools\avr\etc>..\bin\avrdude.exe -c usba
sp -p t85 -B 46.88

avrdude.exe: set SCK frequency to 16000 Hz
avrdude.exe: warning: cannot set sck period. please check for usbasp firmware up
date.
avrdude.exe: error: program enable: target doesn't answer. 0
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

avrdude.exe done.  Thank you.
```

Savannah avrdude ->

```
  -?                         Display this usage.

avrdude version 6.3, URL: <http://savannah.nongnu.org/projects/avrdude/>

C:\tmpavr>avrdude.exe -c usbasp -p t85 -B 46.88

avrdude.exe: set SCK frequency to 16000 Hz
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude.exe: Device signature = 0x1e930b (probably t85)

avrdude.exe: safemode: Fuses OK (E:FF, H:DF, L:62)

avrdude.exe done.  Thank you.
```

per1234, 2017-02-03:
Looks like it could be https://github.com/arduino/avrdude-build-script/issues/1

Please do the following:
1. Plug in the USBasp to your computer.
2. Start Menu > Control Panel > Device Manager
3. Find the USBasp device and double click on it
4. Click the **Driver** tab
5. Write down the `Driver Provider` and `Driver Version`
6. If you're not already using it, install the libusbK driver. I used [Zadig](http://zadig.akeo.ie/) following the directions here: https://github.com/arduino/avrdude-build-script/issues/1#issuecomment-237159133 but I'm sure that's not the only way.
7. Try using your USBasp with the version of AVRDUDE 6.3 supplied with the Arduino IDE
7. If that fixes the problem please post the `Driver Provider` and `Driver Version` that you were using before you changed to libusbK on the issue: https://github.com/arduino/avrdude-build-script/issues/1. Thanks!

facchinm, 2017-02-07:
As suggested by @per1234 , let's move the conversation to arduino/avrdude-build-script#1 , thanks!

closed 2017-02-07 by facchinm