arduino/Arduino #5945: Build Failure (missing arduino-builder-linuxarm-1.3.24.tar.bz2)

Enverex, 2017-02-03:
Build from GIT is failing as the file arduino-builder-linuxarm-1.3.24.tar.bz2 isn't available for download. The actual lines from the build are:

```
untar-unzip-download-web:
      [get] Getting: https://downloads.arduino.cc/tools/arduino-builder-linuxarm-1.3.24.tar.bz2
      [get] To: /home/bot/packages/arduino-git/src/arduino-git/build/arduino-builder-linuxarm-1.3.24.tar.bz2
      [get] Error opening connection java.io.FileNotFoundException: https://downloads.arduino.cc/tools/arduino-builder-linuxarm-1.3.24.tar.bz2
      [get] Error opening connection java.io.FileNotFoundException: https://downloads.arduino.cc/tools/arduino-builder-linuxarm-1.3.24.tar.bz2
      [get] Error opening connection java.io.FileNotFoundException: https://downloads.arduino.cc/tools/arduino-builder-linuxarm-1.3.24.tar.bz2
      [get] Can't get https://downloads.arduino.cc/tools/arduino-builder-linuxarm-1.3.24.tar.bz2 to /home/bot/packages/arduino-git/src/arduino-git/build/arduino-builder-linuxarm-1.3.24.tar.bz2

untar-unzip-download-local:

untar-unzip-checksum:
     [echo] Testing checksum of "./arduino-builder-linuxarm-1.3.24.tar.bz2"
 [checksum] Could not find file /home/bot/packages/arduino-git/src/arduino-git/build/arduino-builder-linuxarm-1.3.24.tar.bz2 to generate checksum for.
```

I assume someone hasn't uploaded that to the website yet.

cmaglie, 2017-02-03:
Oh thanks, it was uploaded really but as `arduino-builder-arm-1.3.24.tar.bz2`, now I've fixed it.

We renamed it from `arm` to `linuxarm` to simplify the `build.xml` but forget to adjust also the script that do bulds and uploads on the CI server.

closed 2017-02-03 by cmaglie

Enverex, 2017-02-03:
The download is completing now, but it's failing the checksum check.

cmaglie, 2017-02-03:
yes, here the fix (already in master):
https://github.com/arduino/Arduino/commit/057419b08b509844b89af2833fa70b11d8ff9a8d

sorry for the troubles...

Enverex, 2017-02-03:
That was quick. Thanks @cmaglie

commit 057419b0 (2017-02-03): Fixed build (rename arduino-builer-arm to arduino-builder-linuxarm)

Fix #5945