arduino/Arduino #5946: IDE shows Wrong user directory!

Eddhsl, 2017-02-03:
Hi Guys, 
found this error and the ide was now unable to compile even the example..
how to rectify this?

I only found that the user was wrong, but i cant found such user inside pc.. 
already uninstall ide and reinstall back, still the same prob.

Error:
Arduino:1.8.1 (Windows 10), 开发板："Arduino Nano, ATmega328"

C:\Users\鑼规煍寰穃Documents\Arduino\Temp_Test\Temp_Test.ino:1:18: fatal error: DHT.h: No such file or directory

compilation terminated.

exit status 1

This "鑼规煍寰穃" is definitely appear from no where...

Cheers!

facchinm, 2017-02-07:
Hi @Eddhsl , you need to install DHT library from Library manager (or manually from [here](https://github.com/adafruit/DHT-sensor-library)

Eddhsl, 2017-02-07:
Hi,
I did install, and can be found in libraries file, but the error still
appear...
Currently using older laptop and its good. May be my last option is format
it..

Thanks anyway.

facchinm, 2017-02-07:
Oh, I see. Let me reopen the issue and ask you to paste here the complete verbose output of the compilation. Does also the old PC has folders with non-latin characters?

Eddhsl, 2017-02-07:
I believe the non latin character was due to the pc set under chinese language.
Old pc set as english.

facchinm, 2017-02-07:
Ok, that's a hint. I'm moving the issue to https://github.com/arduino/arduino-builder/issues/212.
I'd need you to do some debugging to better understand the issue.

- Paste here your home directory name `C:\Users\...` as you expect to see it
- Open the IDE preferences and enable "verbose" on compilation
- Paste here all the output when the compilation fails.

Thanks a lot

facchinm, 2017-02-08:
Moved to arduino/arduino-builder#212

closed 2017-02-08 by facchinm

Eddhsl, 2017-02-09:
> Ok, that's a hint. I'm moving the issue to arduino/arduino-builder#212

Hi,

Well noted.
I will do it once I reach home later.
Hope the error could give a hint on how to solve it before I format it.

Thanks