arduino/Arduino #5949: missing documentation: how to include new 3rd-party libs manually?

shiftleftplusone, 2017-02-05:
hello,
I can't find a documentation about  how to include completely new libs manually.
About libaries there is just 1 topic to be found, but that's about writing own libraries.

http://playground.arduino.cc/Code/Library

e.g., there are many Scheduler Libs, but in the Arduino IDE no single one of them is listes beyond 
Sketch/include library/,
neither in the section "Arduino" , nor in "recommended", nor in "contributed"
and also it's not listed by 
Sketch/manage library/

I also don't want to add a zip lib automatically because it's too opaque, too obscured what's happenning then and whre it's stored and how it may be removed or replaced.

So where is the final guide which tells in which folder the libs have to be copied?

Although
c:\programs(x86)\Arduino\libraries\
is commonly known - which alternatives do exist, e.g. especially for IDE-update-safety purposes?

How to create a proprietary folder which contains all proprietary libs which one once  has picked up and gathered together and which are designed to work also for all Arduino IDE updates in the future, even after complete IDE uninstall and new install with the Windows installer or the non-admin zip container from the scratch?

Where is a topic he references which explain all that?
Nothing to be found by the search in the Playground or in the References.

per1234, 2017-02-05:
See the "Manual Installation" section of https://www.arduino.cc/en/Guide/Libraries.

However, I must say those instructions are just terrible:
- They don't account for the fact that the user might not be using the default sketchbook location.
- They don't account for the 1.5 library format. In fact just today I helped someone on the forum who was trying to apply those instructions to a manual installation of a library in 1.5 format and they were very confused that the official documentation was completely wrong.

shiftleftplusone, 2017-02-05:
I do not even see something about 1.5 library format or either other one or what's the difference anyway!

cmaglie, 2017-02-06:
The place to add your own libraries is `SKETCHBOOK/libraries/YourLibrary/...`.

FWIW when you install a library with the library manager or "from zip", it will be unzipped the path above.

cmaglie, 2017-02-06:
@akash73 the section "Manual Installation" of https://www.arduino.cc/en/Guide/Libraries should be revised with @per1234 suggestions, also the procedure described is a bit confusing.

shiftleftplusone, 2017-02-06:
I have no Sketchbook folder, I am using 3 different USB drives with 3 different paths  randomly.
So a dedicated user-defined path on c:\ or d:\ is needed!

cmaglie, 2017-02-06:
> So a dedicated user-defined path on c:\ or d:\ is needed!

Set your sketchbook folder to that path from preferences

shiftleftplusone, 2017-02-06:
aha!  
So if the path is
d:\libs
will I have to set
d:\libs\
or
d:\libs\libraries\
:?:

cmaglie, 2017-02-06:
if you set `D:\mylibs` you should put the libraries into `D:\mylibs\libraries\...`.
Basically you're using `D:\mylibs` as sketchbook, a bit weird but it should work.

shiftleftplusone, 2017-02-06:
I see,  that was the missing information!
Thank you very much!

akash73, 2017-02-10:
hi all,
page reworked and published.
thanks for notice.

closed 2017-02-10 by akash73

shiftleftplusone, 2017-02-10:
thank you very much, very comprhensible indeed.

> Arduino libraries are managed in three different places: inside the IDE installation folder, inside the core folder and in the libraries folder inside your sketchbook. The location of your sketchbook defaults to “..\<you>\DOCUMENTS\Arduino” on Windows machines. The path is clearly shown in the Preferences panel. If you want to see where the sketchbook defaults open Preferences under File. 

> 

Just 1 question:
where is the core folder?