arduino/Arduino #5953: IDE 1.6.13 Optiboot Makefile Toolroot error

Les-A, 2017-02-05:
Tried to run make for optiboot in ... hardware/arduino/avr/bootloaders/optiboot but makefile line 52 says:
TOOLROOT = ../../../tools
and it should be 
TOOLROOT = ../../../../tools
Results in bash error: avrdude not found.

Other documentation leads me to believe the 'avr' folder is new in the folder structure (just a guess).

Les-A, 2017-02-05:
FYI, just installed IDE 1.8.1 and situation described above still exists.

facchinm, 2017-02-07:
Bootloaders in that folder are there for reference; they only get updated when we ship a new one, so they are likely to break. In this case, the paths are wrong but the parameters are wronger :smile:
If you fix the path you'll get 
```avr-gcc: error: unrecognized command line option '-mshort-calls' ```
which was solved in mainline optiboot a long time ago
 https://github.com/Optiboot/optiboot/issues/101 .
So, even if we fix these problems, we cannot assure that the output binary would work.
We strongly suggest for new projects to use the mainline projects which are maintained or, in case the older version is needed, use the same tools that were present at the time of hex creation.

closed 2017-02-07 by facchinm

Les-A, 2017-02-07:
> Optiboot/optiboot#101 <https://github.com/Optiboot/optiboot/issues/101> .

Thank you for the speedy response.  I'm sure this isn't the place but can
you point me to some reliable instructions on how to get the optiboot
loader into a Nano so I can get a working watch dog?

Many thanks.

facchinm, 2017-02-07:
The optiboot wiki should contain everything needed :wink: https://github.com/Optiboot/optiboot/wiki