arduino/Arduino #5955: Issues with parenthesis ( ) in library names

cmaglie, 2017-02-06:
As discovered in #5935, library names can not have parenthesis in their names.
The problem is that the library manager accepts extra characters like `(` or `)` but once the library is installed it wont be accepted by the IDE with a warning like:

[image]

To fix this the same checks that the IDE does for the library must be applied to the validator of the library manager.

At the moment there are two libraries affected by this problem:

- Open Pixel Control (OPC) Server:
https://github.com/plasticrake/OpcServer
@plasticrake: I see that you alread changed the name to "OpcServer", if it's ok for you I'll change the name to OpcServer also in the index of the library manager.

- Serial Controlled Motor Driver (SCMD)
https://github.com/sparkfun/SparkFun_Serial_Controlled_Motor_Driver_Arduino_Library/tree/V_1.0.0
@marshalltaylorSFE: you must change the name in the library.properties and fix the release (or make a 1.0.1). In the meantime I'll remove the library from the index, since it's unusable anyway in the current state.

cmaglie, 2017-03-02:
- OpcServer: fixed
- SCMD: reported upstream

closed 2017-03-02 by cmaglie

marshalltaylorSFE, 2017-03-02:
Parenthesis removed!
New tagged library:
https://github.com/sparkfun/SparkFun_Serial_Controlled_Motor_Driver_Arduino_Library/tree/V_1.0.1