arduino/Arduino #5958: Add Phpoc to Library Manager

phpoc, 2017-02-07:
This library is to use [PHPoC (Wi-Fi) Shield](http://www.phpoc.com/phpoc_shield_for_arduino.php).

https://github.com/phpoc/arduino

Thank you.

agdl, 2017-02-07:
I'm sorry, but your library can't be added since a tag is missing with the library.properties file inside the tag

phpoc, 2017-02-08:
I am sorry but I couldn't quite catch what you mean.
Our library has [the tag information](https://github.com/phpoc/arduino/tags).
I am not familiar with a "Library Manager" system.
Would you please explain in full detail.

phpoc, 2017-02-08:
I changed the tag information of our repository.
Would you please check it?

agdl, 2017-02-08:
Is not right yet. https://github.com/phpoc/arduino/blob/v1.2.0/library.properties

version=0.8.0 
tag 1.2.0

Fix this please!

phpoc, 2017-02-09:
Finally, I got it.
I fixed the issue related to revision. And I also changed the revision to v1.0.0.
Would you please check it again?
https://github.com/phpoc/arduino/blob/v1.0.0/library.properties

agdl, 2017-02-09:
Done thank you!

closed 2017-02-09 by agdl

phpoc, 2017-02-10:
Thank you so much. ^^