arduino/Arduino #5961: bluetooth rfcomm# does not show up as port for Serial Monitor in linux

linuxgeek, 2017-02-07:
Ubuntu Linux 14.04.2
bluetooth rfcomm# does not show up as port for Serial Monitor in arduino 1.8.1

It does not up under Tools->Port. Only /dev/ttyACM0 shows up.

It's not just a refresh problem, as connecting /dev/rfcomm0 prior to launching arduino 1.8.1.
It does show up as a default board listed on the bottom right such as:
Arduino Yun on /dev/rfcomm0

It won't show up as an available serial port if I connect rfcomm0 after launching arduino 1.8.1

/dev/rfcomm0 still shows in the bottom right of arduino even if I delete /dev/rfcomm0 before launching arduino.

I am using teensy 3.2 device, but even before installing teensyduino, this is the behavior.

linuxgeek, 2017-02-07:
Attempted to switch out liblistSerialsj.so from 1.6.8 created an exception:
Exception in thread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" java.lang.UnsatisfiedLinkError: processing.app.Platform.listSerialsNative()[Ljava/lang/String;
	at processing.app.Platform.listSerialsNative(Native Method)
	at processing.app.Platform.listSerials(Platform.java:180)
	at cc.arduino.packages.discoverers.serial.SerialBoardsLister.retriggerDiscovery(SerialBoardsLister.java:67)
	at cc.arduino.packages.discoverers.serial.SerialBoardsLister.run(SerialBoardsLister.java:183)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)

facchinm, 2017-02-08:
Hi @linuxgeek ,
thanks for spotting this, it looks like libSerialPort is ignoring any serial port with `virtual` in its sysfs path (https://github.com/arduino/libserialport/blob/16e35cfa29ecf4292f19f07141c6b112a6848c69/linux.c#L221) . `rfcomm` ports are indeed virtual but should be listed nevertheless. I'll take a closer look to this issue as soon as I can setup a reliable (and reproducible) rfcomm setup

facchinm, 2017-02-09:
Ok, I tested with some real hardware (a Bluetooth GPS receiver) and `/dev/rfcomm0` is correctly listed in the ports list. It only appears while the device is paired, so maybe it's the problem you are encountering.
About the other "issues":
- the port on the bottom right is only updated when you change it explicitly, and doesn't change when the previously selected port is unavailable.
- `liblistSerialsj.so ` needs to be in sync with the IDE's jars, otherwise you'll get unresolved functions. However, if you want to hack it, the sources are here https://github.com/arduino/listSerialPortsC/

I'm closing the issue as "invalid" since I'm seeing the expected behaviour, please reopen if needed

linuxgeek, 2017-02-09:
I still have the problem. I am definitely paired, and I can communicate with an HC-05 bluetooth receiver.
In 1.6.8 it shows when paired, but doesn't list it when it's not paired.
1.8.1 never shows.

I tried logging in as a different user, and it didn't make any difference. I also tried running as "sudo", no difference.

You're right that /dev/rfcomm0 goes away in the right corner if I explicitly link to /dev/ttyACM0.
Odd though, that in 1.8.1 that it would show /dev/rfcomm0 in the lower right corner, although it has never showed /dev/rfcomm0 in the list of ports in the menu.

The only thing that I can think of that is unusual, is that I use my own script to bring bluetooth up or down. But again 1.6.8 works fine. script below

#!/bin/sh

echo "Usage: <program> <# of rfcomm device>"
RFCOMM="$1"

rfkill block bluetooth
rfkill unblock bluetooth

sudo service bluetooth restart
sudo rfcomm connect "$RFCOMM"

linuxgeek, 2017-02-10:
I don't see how I re-open. Do I just submit another one?

facchinm, 2017-02-10:
I'm reopening it for you.
About the port label on the bottom right, it is due to the IDEs sharing the preference file, so the last used port gets saved and displayed at the next startup (even though the IDE doesn't see it). I'm testing your setup and report here my findings

palto42, 2017-10-08:
I have also issues with the /dev/rfcomm0 not listed in the Arduino IDE. I'm using a HC-05 bluetooth module and tested the IDE versions 1.8.1 and latest 1.8.5 on Ubuntu 16.04.1).
One interesting behaviour I found is that it shows up if I start a gtkterm session to the rfcomm0 port: `gtkterm -p /dev/rfcomm0`. As soon as I close the gtkterm, the port disappears in the Ardunio IDE .
I can also keep the rfcomm0 connection open with the Arduino Serial Monitor.
What I can see in the Ubuntu Bluetooth settings is that the HC-05 is coupled but not connected until I open the gtkterm on port /dev/rfcomm0. If I close gtkterm (and the Arduino Serial Monitor), the connection is immediately switched off again.
[image]

The same happens if I do a simple `echo "Hello" >> /dev/rfcomm0`, the bluetooth connection is turned on, the "Hello" is sent and then the connection is turned off again.

It seems that the Arduino IDE has an issue with this behaviour, hope someone knows how to fix it.

Jul3k, 2021-02-23:
A workaround that helped for me was to bind the bluetooth adapter not with `rfcomm bind` but with:
`rfcomm connect rfcomm1 98:D3:61:FX:XX:XX`
That way the connection is kept alive and Arduino IDE recognized it without problems.

ger01d, 2021-03-15:
Definitely an annoying bug. My workflow, which works 1 of 10 times is to connect first via `picocom -b 115200 /dev/rfcomm0 ` and then I close the connection and in a time window of 3-5 seconds I upload the sketch via arduino-cli. Sometimes it works sometimes not. 
`arduino-cli upload -v -p /dev/rfcomm0 --fqbn arduino:avr:uno test_sketch.ino`

Without the connection to the port via picocom there will be certainly a response 
`avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00`

With the GUI the port is only visible when the connection via picocom is established. But you cannot upload anything in this state.

With `rfcomm connect` I will also get the sync error. So this workflow doesn't work for me. Did anyone find a reliable solution to this?