arduino/Arduino #5968: Kindly document a way to offer pull requests for www.arduino.cc .

aknrdureegaesr, 2017-02-10:
I have not been able to locate the Git repo that contains the sources for the fine documentation at https://www.arduino.cc/ .

Please kindly document that Git repo someplace, so people from the community can help you with the [documentation problems](https://github.com/arduino/Arduino/issues?page=2&q=is%3Aopen+sort%3Acreated-desc+label%3A%22Component%3A+Documentation%22) via pull requests.

per1234, 2017-02-10:
They made a very good first step towards that goal at https://github.com/arduino/reference-en but then completely dropped the ball. That repository now does more harm than good since it's completely out of sync with the actual reference pages and pull requests/issues submitted are not dealt with and [they haven't bothered to even add a note to that effect](https://github.com/arduino/reference-en/issues/22#issuecomment-164434245). Really a shame.

akash73, 2017-02-10:
the integration Is still ongoing.
in the meantime continue to open issues here in github, we will do our best to keep the pages updated.

closed 2017-02-10 by akash73