arduino/Arduino #5970: Serial port name changes every upload

lorenzoromagnoli, 2017-02-10:
Having a weird and annoying issue with the Serial port. 

Every time I upload a sketch on a board the serial port name changes incrementally.

/dev/cu.usbmodem121
/dev/cu.usbmodem122
/dev/cu.usbmodem123
...

Had this on 'El Capitan' with Arduino1.8.1, on two different mac, both with Arduino101 and Arduino zero.

sandeepmistry, 2017-02-10:
I've also encountered this problem, however I don't think it's an issue with the IDE.

For some reason one of the USB ports on my MacBook doesn't seem to provide enough current, after rebooting and using a powered hub I no longer see the problem.

BallscrewBob, 2017-02-12:
Agree with Sandeep.
I can recreate on windows also with lower power USB ports leading to incremental COM port counting.
Macs are known for these types of issue. Adding a POWERED USB 2.0 hub will also overcome most USB 3.0 issues.

lorenzoromagnoli, 2017-02-22:
Ok, then I guess I can close this issue as it is not related to the IDE.

closed 2017-02-22 by lorenzoromagnoli