arduino/Arduino #5976: Separate AVR core into its own repository

mjbcopland, 2017-02-13:
I'd like to be able to fork the AVR core without also getting the IDE, build tools, etc. At the moment this consists of forking the entire Arduino repository and ignoring everything outside of `hardware/arduino/avr`.

Does the AVR core exist as a standalone repository like the [SAM](https://github.com/arduino/ArduinoCore-sam) and [SAMD](https://github.com/arduino/ArduinoCore-samd) cores? If not, could an ArduinoCore-avr repository be created for this purpose? The new repository could be added as a submodule to maintain the current Arduino repository.

facchinm, 2017-02-13:
As soon as we'll process all the PRs related to the AVR core (which are ~30 at the moment) and tag all the relevant issues (a couple of hundreds) we can move the avr core to its own repository. Maybe a great cleanup day could speedup this task a lot :wink: @cmaglie

NicoHood, 2017-02-14:
I'd like to also give some feedback about this:
It would be much better if the core files can also be moved into multiple libraries. For example the Serial library could be split into HardwareSerial and USBSerial.

And then there should be an option to NOT include those. This would safe some overhead and you can overwrite libraries such as the hardwareserial with your very own library. So in the end the "core" would be just a few includes of all standard libraries. This way the IDE is way more modular.

pedzed, 2017-04-15:
> As soon as we'll process all the PRs related to the AVR core (which are ~30 at the moment)

_100 pull requests later..._

Oh hey, it's 129 now.

Personally, I'd just start the repo split process and ask people to resubmit their pull requests to their respective repository. If they are so nice to take the time to submit a PR, this shouldn't be a big deal for them either, I would say.

NicoHood, 2017-04-23:
That is true. And the information is not lost as github shows the old diff.

neu-rah, 2017-04-23:
issues become more easy to track

Rantanen, 2017-04-30:
This would be extremely useful for projects that strive for Arduino compatibility when it comes to libraries and coding but do not need the IDE.

I'm turning my Raspberry PI Zero W into an embedded development platform, which ideally would be able to compile Arduino-compatible code and upload that to Arduino-compatible boards.

I prefer downloading development dependencies over git instead of compressed packages as that allows better control over versioning and easier/faster updating. Unfortunately, when it comes to Arduino, using this official repository is out of the question thanks to the 1.3 gig size.

I worked around this by splitting and hosting the repository on my own at https://github.com/devberry/avr-core. The final repo size came down to 7.0 MB.

carlosperate, 2017-05-01:
>  Unfortunately, when it comes to Arduino, using this official repository is out of the question thanks to the 1.3 gig size.

You can always use the git `--depth` flag.

Rantanen, 2017-05-01:
I've never liked using the `--depth` option for anything but throwaway clones for various reasons. Especially when I'm cloning the repo for someone else through scripts.

It just makes everything behave in an unexpected manner and has [surprising performance characteristics](https://github.com/cocoapods/cocoapods/issues/4989#issuecomment-193772935) if not taken into account everywhere.

I'll stick with my fork workaround for now.

facchinm, 2017-10-11:
AVR core has been moved out of tree in 1.9.x beta branch, base repository becomes https://github.com/arduino/ArduinoCore-avr. Now it's time to move the related issues/PRs :slightly_smiling_face:

sandeepmistry, 2019-09-16:
This has been done :)

https://github.com/arduino/ArduinoCore-avr

closed 2019-09-16 by sandeepmistry