arduino/Arduino #5978: IDE 1.8.1 not compiling any sketches

bukow1cm, 2017-02-13:


bukow1cm, 2017-02-13:
Mac 10.7 OS   Arduino Yun wifi connected to home network ( Yun to my network).  Drivers all up to date.

facchinm, 2017-02-13:
Hi @bukow1cm , could you attach the verbose output of one of the failed compilations? Which Mac model are you using?

SteveSProg, 2017-02-19:
I too have this problem. I'm running under Windows XP, and when trying to verify or compile and download, get a failure message 'arduino-builder.exe has encountered a problem' etc. Investigating this further gives loads of data.  3 screen captures are attached detailing some of this.
[image]
[image]
[image]

facchinm, 2017-02-20:
Hi @SteveSProg ,
we can't extract any useful info from the screens you posted. It would be better to know:
- your PC configuration (processor, ram size, service pack version of your XP)
- a copy/paste of the verbose output of a random compilation (you can toggle it from "preference" menu)

Thanks

SteveSProg, 2017-02-20:
OK.

Error messages:

Arduino: 1.8.1 (Windows XP), Board: "Arduino/Genuino Uno"

D:\Program Files\Arduino\Arduino-1.8.1\arduino-builder -dump-prefs -logger=machine -hardware D:\Program Files\Arduino\Arduino-1.8.1\hardware -tools D:\Program Files\Arduino\Arduino-1.8.1\tools-builder -tools D:\Program Files\Arduino\Arduino-1.8.1\hardware\tools\avr -built-in-libraries D:\Program Files\Arduino\Arduino-1.8.1\libraries -libraries D:\My Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\DOCUME~1\Steve\LOCALS~1\Temp\arduino_build_334617 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=D:\Program Files\Arduino\Arduino-1.8.1\hardware\tools\avr -prefs=runtime.tools.avrdude.path=D:\Program Files\Arduino\Arduino-1.8.1\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=D:\Program Files\Arduino\Arduino-1.8.1\hardware\tools\avr -verbose D:\Program Files\Arduino\Arduino-1.8.1\examples\01.Basics\Blink\Blink.ino

Error compiling for board Arduino/Genuino Uno.

Screen print of my PC configuration: attached

Hope this helps clarify the problem.

Steve

SteveSProg, 2017-02-20:
Hi

Further to my last reply, I checked Arduino.org, which specified that Java 8 was reuired for 1.8.1. Finding I had only Java 7, I loaded 1.8.0 which is stated as requiring Java 7 or later, and still got the same problem, so it would appear to not be that.

Steve

facchinm, 2017-02-21:
@SteveSProg Java is not surely the problem since the builder is written in Golang. It looks you didn't paste the PC specifications, could you try to resend them? Thanks

SteveSProg, 2017-02-21:
Attached as requested

per1234, 2017-02-21:
@SteveSProg email attachments don't work. You either need to include the specs as text in your email response or do the attachment on GitHub.

facchinm, 2017-02-22:
@SteveSProg , no problem, I got the CPU model for the first screenshot (sorry for not trying earlier). It is indeed a Sempron 2400+, so it lacks SSE2 instructions. It seems that https://github.com/arduino/Arduino/issues/4251 and https://github.com/arduino/Arduino/issues/4742 are back, in some way. I'm attaching a builder build on another system, if you could replace the `arduino-builder.exe` binary in `D:\Program Files\Arduino\Arduino-1.8.1` with the one provided and report if it works on your PC it would be great. Thanks!

[arduino-builder.zip](https://github.com/arduino/Arduino/files/792773/arduino-builder.zip)

SteveSProg, 2017-02-22:
Hi.

Looks good, ran Verify on our old favourite 'Blink', and it ran to the point where it shows 'Linking...', then it failed : -

Linking everything together...
"D:\Program Files\Arduino\Arduino-1.8.1\hardware\tools\avr/bin/avr-gcc" -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "C:\DOCUME~1\Steve\LOCALS~1\Temp\arduino_build_16860/Blink.ino.elf" "C:\DOCUME~1\Steve\LOCALS~1\Temp\arduino_build_16860\sketch\Blink.ino.cpp.o" "C:\DOCUME~1\Steve\LOCALS~1\Temp\arduino_build_16860/core\core.a" "-LC:\DOCUME~1\Steve\LOCALS~1\Temp\arduino_build_16860" -lm
collect2.exe: error: ld returned 5 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

I think this is a different problem, having solved the original one.

Steve

facchinm, 2017-02-23:
Ok, we are one step further the resolution. The problem with `ld returned 5` should be solved with https://github.com/arduino/Arduino/pull/5863, so you can download the nightly build and test it (make sure you overwrite `arduino-builder.exe` with the "new" one once you test the nightly)

cmaglie, 2017-03-30:
Fixed on IDE 1.8.2 (and AVR core 1.6.18).

closed 2017-03-30 by cmaglie