arduino/Arduino #5980: Adding the latest hardware source files to Arduino IDE

MCUdude, 2017-02-13:
Hi!
The ATmega48pb/88pb/168pb/328pb and ATmega324pb have been out for more than a year, but Arduino IDE is still missing the source file for these. I'm planning to add proper Arduino support for these, but nobody are going to use them if the source files are missing. How can these be added? I downloaded the latest version of Atmel Studio 7, and all the necessary source files can be found within that package.

How can these be added to the Arduino IDE? I think getting support for these new puppies will bring alot of new and interesting HW to the market 😃

matthijskooijman, 2017-02-14:
What do you mean with "source files" exactly? Do you mean that the toolchain (compiler/linker/uploader) does not know about them, or do you mean that the Arduino core does not support these?

For the former, a toolchain update is needed. I think Arduino usually tracks Atmel Studio (which has a bunch of patches), but I'm not sure if there's any updates there that aren't present in the Arduino IDE yet (Atmel Studio isn't very fast with toolchain updates I think).

For the latter, code changes are needed. Since there are no official Arduino boards using those MCUs AFAIK, I suspect that the Arduino team won't be investing time into these, but clean pull requests for support will probably be merged.

MCUdude, 2017-02-14:
I'm talking about the source files for the toolchain and avrdude. I know which directories the corresponding source files should be located on my computer, but I can't find these directories anywhere on github!

Here's an Arduino core for the new ATmega328PB. The instructions contains a link to the source files that needs to be installed manually.I want to create a pull request where I add these files, but I can't find the directories on github. Where is the io.h file located? Where is the avrdude.conf file?

Thanks!

matthijskooijman, 2017-02-14:
The toolchain files are downloaded and unpacked during the build, so they are not stored in this repository. The scripts that build these downloaded files are here, so that's where changes need to happen: https://github.com/arduino/toolchain-avr

However, the toolchain files are always compiled from source, from upstream versions, so just patching a few files to support these chips is not going to happen.

It seems there is already an open issue in that repo, so I'm closing this one in favor of that one: https://github.com/arduino/toolchain-avr/issues/36

closed 2017-02-14 by matthijskooijman