arduino/Arduino #5984: Automatic function prototype generation does not work inside namespaces

nickgammon, 2017-02-13:
In the example code below I am placing some functions into a namespace to avoid clashes. (This is in answer to a [question about combining two sketches](http://arduino.stackexchange.com/questions/34497/). However the automated function-prototype generation seems to be suppressed inside a namespace.

```c++
namespace foo 
{
//  void bar ();
  
  void fubar ()
     {
     bar (); 
     }  // end of fubar
     
  void bar ()
    {
    }  // end of bar
}  // end of namespace foo 

void setup()
{
  foo::fubar ();
}  // end of setup

void loop()
{
}  // end of loop
```

If you uncomment the manual function prototype near the top the code compiles OK. As written however I get the error:

```
/home/nick/Pictures/Blink tutorial/Example/Example.ino: In function 'void foo::fubar()':
Example:7: error: 'bar' was not declared in this scope
      bar (); 
           ^
exit status 1
'bar' was not declared in this scope
```

This is with IDE 1.8.1.

facchinm, 2017-02-14:
Hi Nick, prototypes are not autogenerated for classes and namespaces (see https://github.com/arduino/arduino-builder/blob/a4ede088e7f5fa67fd7c2f12b94e3124126bae00/src/arduino.cc/builder/ctags/ctags_parser.go#L206) . The reason is that it's really difficult to find the correct insertion point for these prototypes, so we avoid creating them (silently). Maybe [this](https://github.com/arduino/arduino-builder/issues/167) proposal could help? Would you mind continuing the discussion there?

nickgammon, 2017-02-15:
OK, I'll close this issue as it seems to be by design.

closed 2017-02-15 by nickgammon