arduino/Arduino #5987: Arduino studio crashes on press of combination: Shift + O (not a zero but a o)

zigge, 2017-02-14:
When the IDE starts up, it works like a charm, but as soon as the combination of shift + o is pressed it crashes. It says that Java SE has stopped working, and it closes the program. 

Specs: Windows 10 EDU. 
Java SE: 1.8.0_121
Arduino Studio: 1.8.1

matthijskooijman, 2017-02-14:
Can you try running arduino_debug.exe from a commandline prompt, to see if any extra error output is given when it crashes?

zigge, 2017-02-14:
I've done that already, and there is no extra output from the console.. It's like the debug program does not catch any errors or exceptions.

matthijskooijman, 2017-02-14:
Weird. Is there any extra info to be extracted from the crash message? I'm really puzzled about what happens here, since shift+O is supposed to just type a capital letter O, there shouldn't be anything special happening then.

Does it happen always, even when the main text area is not focused (but the compilation output window, for example), and with any sketch loaded?

zigge, 2017-02-14:
Hmm, no there is not much other than what i have described already :/ 
Yes the shift + o is really weird. It happens no matter if the main sketch is in fokus or not.
[image]

This is how it shows ^

dlabun, 2017-02-18:
By any chance are you using a specialty keyboard like Logitech gaming keyboards?