arduino/Arduino #5988: issue about Scheduler (mpatie) vs. Scheduler (cmaglie)

shiftleftplusone, 2017-02-14:
hi,
I just run into this issue with my Due:

the former code using cmaglie's Scheduler works fine,
but with mpatel's Scheduler the program freezes after
 curlf();   sprintf(sbuf, " ...", _curx_); lcdprint(sbuf);   delay(333);   // <<<<<<<<<<<<<< 

what might be the reason?

`/*   Arduino Remote Console     
 *   UART / Bluetooth HC-05 (master)
 *   Arduino Due
 *   IDE 1.6.5
 *    
 *   version RCsender0013
 *  
 * change log: 
 *   "heartbeat"
 *   
 */
 
// (C) Helmut Wunder (HaWe) 2015
// freie Verwendung für private Zwecke
// für kommerzielle Zwecke nur nach Genehmigung durch den Autor.
// Programming language: Arduino Sketch C/C++ (IDE 1.6.1 - 1.6.5)
// protected under the friendly Creative Commons Attribution-NonCommercial-ShareAlike 3.0 Unported License
// http://creativecommons.org/licenses/by-nc-sa/3.0/

 
#include <SPI.h>
#include <Scheduler.h>

// SPI header            //             |--|
// SPI pins Due          //  -----------|  |------------
  #define DUEMISO     74 //  |  RES   76_SCK   74_MISO |
  #define DUEMOSI     75 //  | -GND   75_MOSI   Vc+5V  |
  #define DUESCK      76 //  --------------------------

// SPI pins Mega
//#define MISO     50
//#define MOSI     51
//#define SCK      52

//=====================================================================================
// TFT LCD
//=====================================================================================

#define   UTFT_SmallFont     8 // UTFT 8x10
#define   UTFT_MediumFont   12 // UTFT ++
#define   UTFT_BigFont      18 // UTFT +++
#define   _SmallFont_        1 // 9341 6x9
#define   _MediumFont_       2 // 9341 12x16
#define   _BigFont_          3 // 9341 18x23

int16_t  LCDmaxX , LCDmaxY ;                // display size
int16_t  _curx_, _cury_,                    // last x,y cursor pos on TFT screen
         _maxx_, _maxy_;                    // max. x,y cursor pos on TFT screen       

// set LCD TFT type
int16_t  LCDTYPE   =   -1;

#define  _LCD1602_    1  // LCD1602  Hitachi HD44780 driver <LiquidCrystal.h>
                           // http://www.arduino.cc/en/Tutorial/LiquidCrystal   //
#define  _UTFT_       4  // Henning Karlsen UTFT 2.2-2.4" 220x176 - 320x240 lib
                           // http://henningkarlsen.com/electronics/library.php?id=51   //
#define _ILI9341_     8  // https://github.com/adafruit/Adafruit_ILI9340
                           // https://github.com/adafruit/Adafruit-GFX-Library //
#define _ILI9341due_  9  // ILI9341_due NEW lib by Marek Buriak
                           // http://marekburiak.github.io/ILI9341_due/ //
                           
[... 664 lines cut ...]
   curlf(); 
   lcdprint(sbuf);
   Scheduler.startLoop(loop3); 
   sprintf(sbuf, "MT heartbeat task started "); 
   curlf(); 
   lcdprint(sbuf);
   curlf(); 

   sprintf(sbuf, "setup(): done.");
   Serial.println(); Serial.println(sbuf);   
   curlf(); curlf(); lcdprint(sbuf);

   sprintf(sbuf, "press JoystickButton 0 to start!");
   Serial.println(); Serial.println(sbuf);   
   curlf(); curlf(); lcdprint(sbuf);

   while( !sensortouch( joystBtn_0 ) );   while( sensortouch( joystBtn_0 ) );
   lcdcls(); 
   curlf();   sprintf(sbuf, " ...", _curx_); lcdprint(sbuf);   delay(333);   // <<<<<<<<<<<<<< 
     
   lcdcls(); 

}

`

facchinm, 2017-02-14:
I believe this should be reported to https://github.com/mikaelpatel/Arduino-Scheduler, thanks.

closed 2017-02-14 by facchinm

mikaelpatel, 2017-02-14:
I guess that there is a difference in default stack size.

shiftleftplusone, 2017-02-15:
even after changing the initial stack to 1024 the issue remains.

mikaelpatel, 2017-02-15:
@shiftleftplusone You need to change the stack size for the main task as well. Or simply download the new version which has the same default stack as the arduino-library/Scheduler.

shiftleftplusone, 2017-02-15:
I have done all that already

cmaglie, 2017-02-15:
This is unrelated with the Arduino IDE, please move the discussion to the correct issue tracker: https://github.com/mikaelpatel/Arduino-Scheduler