arduino/Arduino #5993: Arduino Due freezes just after declaring a Rtc_Pcf8563 type object in 1.8.1 IDE.

linzym, 2017-02-16:
I think that the wire library is a standard library for all arduino boards. But when I use the wire library on an arduino due board, I find it doesn't work, even if I use an example code and comment out any other statements related with the instance rtc.

I use the code below just for a test:

#include <Wire.h>
#include <Rtc_Pcf8563.h>

//init the real time clock
Rtc_Pcf8563 rtc;

void setup()
{
  //clear out the registers
  //rtc.initClock();
  //set a time to start with.
  //day, weekday, month, century(1=1900, 0=2000), year(0-99)
 // rtc.setDate(14, 6, 3, 1, 10);
  //hr, min, sec
 // rtc.setTime(1, 15, 0);
  Serial.begin(9600);
}

void loop()
{
  //both format functions call the internal getTime() so that the 
  //formatted strings are at the current time/date.
  //Serial.print(rtc.formatTime());
  Serial.print("\r\n");
 // Serial.print(rtc.formatDate());
  Serial.print("\r\n");
  delay(1000);

  //debug
  Serial.write(0x5A);
}

What's the problem with the wire or twi library running on an Arduino Due board?

facchinm, 2017-02-16:
Wire on the Due behaves exactly like Wire on any other architecture. The only difference is that  `Wire` object is mapped on pins 20-21 while `Wire1` is mapped on SCL1-SDA1, so you may need to change the library if you are using the second port. 

I can't test with real hardware, but the `Rtc_Pcf8563` constructor calls `Wire.begin()` so it should be safe.
```
Rtc_Pcf8563::Rtc_Pcf8563(void)
{
	Wire.begin();
	Rtcc_Addr = RTCC_R>>1;	
}
```
Also, make sure you are using the Programming port, or change `Serial` into `SerialUSB` if you prefer the Native one.

linzym, 2017-02-27:
Hi Facchinm,

Thanks for your comments, and so I confirmed that the wire library for Due is OK from your comments. After that, I spent several days to learn the libraries of Arduino and test my Pcf8563 board. When testing, the board is running well. And an experienced engineer told me about directory ‘arduino15’, in which the wire library files of Due board in my computer are stored. So I found I must run the examples in the directory ‘arduino15’ in order to use the wire library for Due. Now, I think the problem is solved. Besides wire library, several other libaries for Due board were rewritten and stored in directory ‘arduino15’. 

Thank you very much!

closed 2017-02-27 by linzym