arduino/Arduino #5994: Library manager, show buttons and dropdowns all the time

septillion-git, 2017-02-16:
One thing that kept me busy for an evening was the fact I missed a library had an update. Although I opened the Library Manager it just shows as "Installed". But only AFTER clicking the library it shows the buttons "install" and more important "update"! Please show them all the time or at least show there is an update (in bright orange or something).

And I'm not the only one who pulled his hair out, also famous YouTuber Julian fell into the same pit. See
https://www.youtube.com/watch?v=LLDdEcoh-oo

So if I, as a advanced programmer and experienced Arduino user makes that mistake I bed all newbies do the same...

per1234, 2017-02-16:
I've never understood what was the purpose of doing it that way. It's very unexpected and non-standard behavior. I've seen multiple tutorials that say you have to click the "more info" link in Boards Manager before you can install a hardware package. They didn't realize you just have to click anywhere on the entry for that package. I remember reading a blog or forum post from a teacher whose entire class was stymied by this strange UI behavior until one student happened to figure it out by chance.

septillion-git, 2017-03-21:
What's the status of this? Is it going to be fixed? It's can't be to hard... 😅

per1234, 2020-12-24:
Although your requested "show all the time" change has not been made, the situation has since been significantly improved in that the buttons now shown on hover, rather than requiring a click on the platform entry.

septillion-git, 2021-01-02:
First of all, best wishes! Sorry for the late reply, the holidays got in the way. Thanks for the update. I never really noticed any difference but on hover is indeed an improvement. 😄 But I still think it makes it unnecessary more complex without advantages. After all, it's just a functional menu.