arduino/Arduino #5999: Python error when compiling for the ESP32 from Arduino IDE 1.8.1

dibaggioj, 2017-02-18:
A python error when calling `dlopen` is occurring when attempting to compile any Arduino sketches (e.g., Blink) for the Sparkfun ESP32 The Thing, per https://github.com/espressif/arduino-esp32.

The verbose log is below:
```Arduino: 1.8.1 (Mac OS X), Board: "SparkFun ESP32 Thing, 80MHz, 115200"

/Users/johndibaggio/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Users/johndibaggio/Applications/Arduino.app/Contents/Java/hardware -hardware /Users/johndibaggio/ESP32/hardware -tools /Users/johndibaggio/Applications/Arduino.app/Contents/Java/tools-builder -tools /Users/johndibaggio/Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Users/johndibaggio/Applications/Arduino.app/Contents/Java/libraries -libraries /Users/johndibaggio/ESP32/libraries -fqbn=espressif:esp32:esp32thing:FlashFreq=80,UploadSpeed=115200 -ide-version=10801 -build-path /var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/arduino_build_247649 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Users/johndibaggio/ESP32/sketches/ESP32-Blink/ESP32-Blink.ino
/Users/johndibaggio/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Users/johndibaggio/Applications/Arduino.app/Contents/Java/hardware -hardware /Users/johndibaggio/ESP32/hardware -tools /Users/johndibaggio/Applications/Arduino.app/Contents/Java/tools-builder -tools /Users/johndibaggio/Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Users/johndibaggio/Applications/Arduino.app/Contents/Java/libraries -libraries /Users/johndibaggio/ESP32/libraries -fqbn=espressif:esp32:esp32thing:FlashFreq=80,UploadSpeed=115200 -ide-version=10801 -build-path /var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/arduino_build_247649 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Users/johndibaggio/ESP32/sketches/ESP32-Blink/ESP32-Blink.ino
Using board 'esp32thing' from platform in folder: /Users/johndibaggio/ESP32/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /Users/johndibaggio/ESP32/hardware/espressif/esp32
Detecting libraries used...
"/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_THING -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/cores/esp32" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/variants/esp32thing" "/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/arduino_build_247649/sketch/ESP32-Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_THING -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/Users/johndibaggio/ESP32/hardware/espressif/esp32/cores/esp32" "-I/Users/johndibaggio/ESP32/ha [... cut ...] vm1cjm0000gr/T/arduino_build_247649/ESP32-Blink.ino.elf"
Error loading Python lib '/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/_MEIUUYsVm/Python': dlopen(/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/_MEIUUYsVm/Python, 10): no suitable image found.  Did find:
	/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/_MEIUUYsVm/Python: code signature invalid for '/var/folders/2d/qmc6fvtn7b96mh0qjnvm1cjm0000gr/T/_MEIUUYsVm/Python'

exit status 255
Error compiling for board SparkFun ESP32 Thing.```

matthijskooijman, 2017-02-18:
This is an error in the `esptool` program, which is not shipped or maintained by the Arduino team. You should probably report this to the repository you linked instead. I'm closing this issue.

closed 2017-02-18 by matthijskooijman

dibaggioj, 2017-02-18:
@matthijskooijman, thanks, I've gone ahead and done that.