arduino/Arduino #6003: arduino IDE porblems

Allen333, 2017-02-20:
I am having problems with sketchbook. In the file menu when I hover over sketchbook it doesn't display the sketch I loaded in using preferences,

facchinm, 2017-02-20:
Hi @Allen333 ,
in the preference menu you should specify the root folder containing the sketches, not the sketch itself.
So, if your folder tree resembles

```
- mysketchbook
- mysketchbook\sketch1\sketch1.ino
- mysketchbook\sketch2\sketch2.ino
- mysketchbook\libraries\Library1\....
```

the correct folder to point to is `mysketchbook`. Note that every sketch must be contained in a folder with the very same name, so if you have a collection of sparse .ino you should put them in the correct folders first.

Allen333, 2017-02-20:
I think I figured out the problem thanks for your help.

cmaglie, 2017-02-20:
Anyway the correct place to ask for user support is the forum:

https://forum.arduino.cc

closed 2017-02-20 by cmaglie