arduino/Arduino #6006: [new library] AccelStepper

waspinator, 2017-02-20:
Please add this library.

https://github.com/waspinator/AccelStepper

Thanks

cmaglie, 2017-02-21:
Done!

closed 2017-02-21 by cmaglie

PaulStoffregen, 2017-02-21:
Was this repository made with Mike's permission or understanding?

waspinator, 2017-02-21:
@PaulStoffregen no, I'm not Mike. I tried to make it that clear with the repo's description and readme. I am only maintaining it for Arduino's library manager. It's a great library, and I wanted to make it easier for others to install. 

I didn't want to bother Mike, as I know he's very busy. I wasn't aware that his permission was required, or that there would even be an issue. If Mike has a concern, I would be happy to talk to him about it.

cmaglie, 2017-02-21:
The library is released under GPL so it doesn't require explicit permissions to fork and the [library.properties](https://github.com/waspinator/AccelStepper/blob/master/library.properties) correctly reports:

```
author=Mike McCauley <mikem@airspayce.com>
maintainer=Patrick Wasp <patrickwasp@gmail.com>
```

anyway, it's good practice to coordinate with the upstream author, at least to make him aware that this fork exists and you wish to maintain it.