arduino/Arduino #6013: localIP() delivers incorrect IP address; RXIP/RXIN correctly wired ?

arduinoFriend, 2017-02-22:
Hi everybody,

I face a similiar problem as @Alienwaren back in 2014 (https://github.com/arduino/Arduino/issues/2168), which I get with the code pasted below. I have a Leonardo ETH.

@PaulStoffregen who found a good solution in 2014 is kindly invited to join the discussion.

The Leonardo ETH also has a SD controller on board, but I have no SD card plugged in. 

I  set a static IP address on the W5500 Ethernet controller. The returned IP by localIP() is not correct (0.178.178.178), which is stable in one session.  It changed a few times on new power ups. When I continue by doing a EthernetClient connect, it fails as could be expected. Setting the controller up by DHCP gives the same behaviour or causes an eternal Ethernet.begin ... 

My assumption is that the problem lies in the SPI communication between the ATMega and the ETH controller or in the ETH controller itself (broken ?).

A second issue concerns the Leonardo ETH board schematics (Arduino_Leonardo_Eth_v1_sch.pdf) and the Ethernet Shield 2.  Nets 'RXIP' and 'RXIN' are AC-coupled (i.e. by capacitors) to the RJ45 jack, but they seem to be absolutely DC-floating. If there is no DC path, this can affect heavily communication reliability and must be considered a bug !

Thanks for every answer,
Wolfgang - new to Arduino but not new to electronics

```
#include <SPI.h>
#include <Ethernet.h>

// MAC address
byte mac[] = { 0x90, 0xA2, 0xDA, 0x10, 0xE4, 0x55 };
// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 0, 9);

void setup() {

  pinMode(10, OUTPUT);    // make sure the the ETH controller is enabled
  digitalWrite(10, LOW);
  
  pinMode(4, OUTPUT); // make sure that the SD card controller is not accidentially outputting on SPI shared with Ethernet
  digitalWrite(4, HIGH);
  delay(1);
  
  // Start serial logging
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Ethernet.begin(mac, ip);
}

void loop()
{
  Serial.println(Ethernet.localIP());
  delay(1000);
}
```

per1234, 2017-02-22:
The Ethernet library included with the Arduino IDE [doesn't support the W5500 Ethernet controller](https://www.arduino.cc/en/Reference/Ethernet) found on the [Leonardo ETH](https://www.arduino.cc/en/Main/ArduinoBoardLeonardoEth). Support requests should be posted to the [Arduino forum](https://forum.arduino.cc/). This issue tracker is for bug reports or feature requests only. I can't comment on the electronic issue mentioned, that may be appropriate for an issue report here.

arduinoFriend, 2017-02-23:
Thanks for the quick and clear answer. I will post a support request at the Arduino forum.

Rotzbua, 2017-02-23:
Btw the library moved to https://github.com/arduino-libraries/Ethernet and there is an experimental with 5500 support. The pr for 5500 support is pending for years, so if you test it may help...

PaulStoffregen, 2017-02-23:
Please give this copy a try.

https://github.com/PaulStoffregen/Ethernet

It supports all chips, and it has major performance improvements.

arduinoFriend, 2017-02-24:
Thanks for the comments. The Leonardo ETH has a W5500 controller and I simply sourced the wrong lib (that is to say the orginal Ethernet library). 

@PaulStoffregen I tried your Ethernet library but it did not detect any ETH controller chip. I got it confirmed "officially" by uncommenting the corresponding line in w5100.cpp. 

@Rotzbua: with neither Ethernet-master nor Ethernet-master experimental I got something working.

Ethernet2 and the Wiz Ethernet library worked. This statement refers to the simple Telnet client example.

Unfortunately in my real program I saw strange and unstable behavior, with the Wiz library behaving a bit better. Sometimes a client could connect to a web site and return the write response, sometimes it connected successfully but client.available remained false. The relevant code I copied to the Telnet example and it worked there ! Could my problems be due to insufficient RAM (the global variables of my program use 1.5kB out of 2.5kB total RAM) ? 

Anyway, since my final program is even larger, I need to think about a MEGA 2560. Even though it is a bit unsatisfactory to make the next step.

Regards,
Wolfgang

arduinoFriend, 2017-02-25:
I may not tell any news but my problems got solved by using the ingenious double read loop 
```
while(client.connected()) {
   while(client.available()) {
```
as mentioned here: http://forum.arduino.cc/index.php/topic,99629.msg754629.html#msg754629
Best regards,
Wolfgang