arduino/Arduino #6020: Version 1.6.10+ breaks .indexOf

2E0PGS, 2017-02-26:
I have a project that uses the Ethernet Shield, Arduino UNO and `.indexOf` function to check the HTTP response for a specific string. This code works fine in older versions of the IDE such as 1.6.8, 1.6.9. However 1.6.10 and above does not function as expected. The console shows that `.indexOf` function is not properly extracting data and causes issues with the source variable it's trying to check against. Here is a snippet of the code for example: 
`else if (HTTP_req.indexOf("button8on") > 0) {
            dataToSend = 9;
}`

I have checked the change logs for 1.6.10 and I see there is alot of changes. The two main ones that stand out to me in relation to this issue are: "* Another small fix to String iterators. Thanks @Ivan-Perez @Chris--A" and "* A lot of bugfixes to builder:
  https://github.com/arduino/arduino-builder/issues?q=milestone%3A1.3.19+is%3Aclosed".

I don't know if this is expected but to me it seems like incorrect behaviour on versions 1.6.10 and up.

Bare in mind I do get a warning for low memory: "Low memory available, stability problems may occur."
I have this warning on 1.6.9 version too and that works just fine.

matthijskooijman, 2017-02-26:
> Bare in mind I do get a warning for low memory: "Low memory available, stability problems may occur."
I have this warning on 1.6.9 version too and that works just fine.

This might very well be the cause of your problem, since the String class uses dynamic memory which is not included in the memory calculation at compile time. Could you try if you can reduce the memory load of your sketch (e.g. by shrinking global arrays, removing libraries, etc.). If that helps, you're likely seeing a memory problem. If that does not help, please reduce your problem sketch to a simple, self-contained sketch that shows the problem and paste that sketch here. Thanks!

SasmitoMotion, 2017-02-26:
how to uninstall arduino-nightly windows mr Matthijs and install 1.6.9

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

2E0PGS, 2017-02-27:
Will try that now. @matthijskooijman

2E0PGS, 2017-02-28:
I have reduced the code size by commenting out some `client.println` under the Ethernet server code. This reduced code which no longer has a high memory usage warning acts as expected. So this does seem to be a memory related issue. That concludes that. 
It seems Ethernet libraries use alot of memory with a large HTML page being sent via `client.println`. I shall have to now try reduce code size while maintaining functionality of my project.

Cheers @matthijskooijman
-2E0PGS

closed 2017-02-28 by 2E0PGS