arduino/Arduino #6024: Is this an IDE bug, a Serial bug or i'm doing it wrong?

snobu, 2017-02-27:
Look at this one - if i don't make the dummy String assignment i can't `Serial.print()` the char array:

```c++
void loop() {
  float myfloat = 28.21;
  char buf[20];
  snprintf(buf, sizeof buf, "%.2f", myfloat);
  Serial.print("buf = "); Serial.println((char*) buf);
  Serial.print("sizeof buf = "); Serial.println(sizeof(buf));

  delay(4000);
}
```

Result:
```
buf = 
sizeof buf = 20
buf = 
sizeof buf = 20
buf = 
sizeof buf = 20
buf = 
```

```c++
void loop() {
  float myfloat = 28.21;
  char buf[20];
  snprintf(buf, sizeof buf, "%.2f", myfloat);
  Serial.print("buf = "); Serial.println((char*) buf);
  Serial.print("sizeof buf = "); Serial.println(sizeof(buf));

  // I'm simply adding in this String assignment,
  // not actually using it
  String dummy = buf;
  
  delay(4000);
}
```

Result:
```
buf = 28.21
sizeof buf = 20
buf = 28.21
sizeof buf = 20
buf = 28.21
sizeof buf = 20
buf = 28.21
```

descampsa, 2017-03-07:
You should use dtostrf instead of snprintf.
`dtostrf(myfloat, 2, 2, buf);`

Float formatting using printf functions is generally disabled in Arduino because it takes a lot of program space.

As for why it works with the String, i guess that it enable compilation of https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/avr/dtostrf.c, and that the line https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/avr/dtostrf.c#L23 enable float support in snprintf.

Not sure if it is intended behaviour or not, but that appears to be specific to the samd platform (the String does not change the behaviour on Uno).

facchinm, 2017-03-09:
Moved to https://github.com/arduino/ArduinoCore-samd/issues/217

closed 2017-03-09 by facchinm