arduino/Arduino #6025: IDE text with high resolution VDU

hauwarea, 2017-02-27:
My PC has a SAMSUNG U28D590D VDU driven by an NVIDIA QUADRO K620 Video Card.  As a result the text size and colour in the Arduino IDE (1.6.13) have become very hard to discern with my aged eyesight.  It would be a great boon to me, and I am sure also to others suffering terminal old age, if there were to be greater control over the typeface and the colours for reserved words, variables, constants etc.

Adding this functionality to the IDE might not win anyone a Nobel Prize for computing, but it would be greatly appreciated by many users.

Thanks

per1234, 2017-02-28:
You can easily do some of this via **File > Preferences > Editor font size** and **File > Preferences > Interface scale**. More advanced customization such as colors and fonts can be done by editing the theme files. Unfortunately there is no official documentation for this that I'm aware of but here's my attempt at documentation and a sketch to make it easy to test theme settings:
https://github.com/per1234/ThemeTest

The ability to easily edit the Arduino IDE's theme is very important for accessibility. This is something reported frequently on the forum. In that interest I suggest the following:
- Publish official documentation for the theme files
- Allow theme files to be put in the Arduino15 folder or sketchbook so theme edits aren't lost when the IDE is updated.

hauwarea, 2017-02-28:
Wow!  Thanks for the effort, but that is way over my head.  I am a mere user, not a developer or even a wannabee developer.  I was hoping someone would take up the challenge of facilitating personalization of the interface with dropdown menus and suchlike through the User Interface itself.  Hard-coding via Theme Files is way beyond my limited skills.

closed 2018-02-13 by hauwarea