arduino/Arduino #6027: Unable to Upload to mBot without opening as Administrator

lukekroebel, 2017-02-28:
Hello Everyone,

The mBlock website keeps pointing me to this forum, so I'm hoping someone can help me.

I'm unable to upload to the mBot I have through the mBlock Program.

I get the following error:

C:\Program Files (x86)\mBlock\Arduino\arduino_debug.exe --upload --board arduino:avr:uno --port COM3 --verbose --preserve-temp-files C:\Users\student2022\AppData\Roaming\com.makeblock.Scratch3.4.5\Local Store\scratchTemp\project_Untitled1_1\project_Untitled1_1.ino
Loading configuration...
org.apache.commons.exec.ExecuteException: Process exited with an error: 1 (Exit value: 1)
at org.apache.commons.exec.DefaultExecutor.executeInternal(DefaultExecutor.java:377)
at org.apache.commons.exec.DefaultExecutor.execute(DefaultExecutor.java:160)
at org.apache.commons.exec.DefaultExecutor.execute(DefaultExecutor.java:147)
at processing.app.windows.Platform.getFolderPathFromRegistry(Platform.java:70)
at processing.app.windows.Platform.recoverSettingsFolderPath(Platform.java:55)
at processing.app.windows.Platform.init(Platform.java:50)
at processing.app.Base.guardedMain(Base.java:153)
at processing.app.Base.main(Base.java:137)

Though, when I open mBlock as Administrator uploading works fine.

Also I used in the past mBlock version 3.0.0 works without a problem.

The problem started with Version 3.3.5 when updating from 3.0.0.

I've given the student user Admin permissions to the Folder for mBlock, but this didn't help.

Education environment.
Windows 10.

I cannot figure out how to fix this issue.

Any help would be great.

Many thanks in advance,
Luke

facchinm, 2017-03-01:
Hui @lukekroebel ,
it looks like the IDE is experiencing troubles while accessing the preferences file. [This](http://forum.arduino.cc/index.php?topic=332952.0) forum entry describes your problem but doesn't give a proper solution. 
Since `getFolderPathFromRegistry` was introduced with f6546aea4527bb0723fac6472c1e10328ebe9473 (before 1.6.5) and removed before 1.6.6, so I strongly suggest to update your IDE installation before trying any other solution :slightly_smiling_face:

lukekroebel, 2017-08-04:
Hey

closed 2017-08-04 by lukekroebel