arduino/Arduino #6032: platform.txt for a specific sketch

Wurstnase, 2017-03-02:
Hi all,

I'm working on a project which needs some special flags for the compiler. While this could be add in a the platform.local.txt or platform.txt, it would be great if someone could add a platform.sketch.txt or similar in the sketch-folder, which is also used for compiling.

Whit that, we could add compiler options individually for each project.

Or are there any other ideas to add individual flags for the compiler for a specific sketch?

facchinm, 2017-03-02:
Hi @Wurstnase ,
the feature you are requesting has been proposed multiple times (see https://github.com/arduino/Arduino/issues/5084#issuecomment-229494798 for a sum-up) but was always rejected since we know that developers' imagination is endless and we would end up with very subtle and non reproducible bugs. However, sooner or later we could adopt a soft version of one of these proposal, leaving the opportunity to change small bits of the platform configuration at runtime.

Wurstnase, 2017-03-02:
Hi @facchinm,
Not the answer I want to hear, but thank you for the quick response.

closed 2017-03-02 by Wurstnase