arduino/Arduino #6034: Disabling unused boards in the IDE

DCEM, 2017-03-03:
Hi, 

this is a feature-request. I hope I'm doing this the propper way.

I would like to be able to tidy up the Arduino IDE by removing all the boards that I actually don't own/use.

They still should be there and update - I just would like for them not to show up.

Best regards
DCEM

PaulStoffregen, 2017-03-03:
Agreed, some of the 3rd party packages add a tremendous number of boards.

matthijskooijman, 2017-03-06:
I've had a different solution to that problem in mind: Adding a "recently used boards" list at the top or in a submenu, that shows the 10 or so most recently used boards for quick selection. That's probably a lot easier to implement and should pretty much solve the same problem I think?

DCEM, 2017-03-09:
The problem with a "recently used" menu is that there are going to be people that use more boards than you will be able to find there.

Also you are going to have these entries doubled - witch wouldn't be more tidy from my point of view.

I would prefer a blacklist.

kf2402, 2017-06-05:
Rather a whitelist, stored in txt or json among the users personal stettings, which can ideally be edited through the board manager.

per1234, 2018-10-23:
There is an undocumented feature that allows you to hide boards by adding a `hide` property to boards.txt. For example, in Arduino AVR Boards the Arduino/Genuino Uno board can be hidden by adding this:
```
uno.hide=
```
It doesn't matter what value you set the property, just the presence of that property causes the board to not be shown in the **Tools > Board** menu. This isn't an ideal solution because it means you would need to redo the modification after every hardware package update. That can be made slightly easier by putting the `hide` property definitions in a boards.local.txt file.

stef-ladefense, 2019-01-29:
good idea the white list but how to implement it easily? And especially in the directory of the local user, that the change is not lost by installing a later version.
It would be necessary to have the list of all the cards installed and recognized by the IDE, and to be able to check it or uncheck them via the interface to hide some of them.
But it's true that it would be really nice to display only the cards actually used.

Subsidiary questions: it would be necessary to be able to easily configure the default options of the cards used, not very useful for the standard cards, but for the ESP it is with each change of card that it is necessary to reconfigure the options of speeds, sizes, etc. that we change the map, for example between a Lolin8266 D1 and a uno, or a tiny, and when we come back to the Lolin, we lost the config that we put on it.

hasenradball, 2019-04-24:
You explaned the option for hide... where has the hide option placed just for the name?

klavatron, 2019-12-08:
Maybe its better to add "Favourite Boards" submenu (like "Boards") and menu item "Add/Remove to Favourites" (Like "Get board info") instead of disabling unused boards

hasenradball, 2019-12-08:
Boards Favorite would be cool!

henrygab, 2020-03-20:
Board Favorites -- This is so much better than blacklist / hiding individual boards.  After all, none of these are convenient:
* initial addition of all non-desired boards to a boards.local.txt
* reviewing which boards are hidden (split across multiple package directories...)
* enabling a board which was previously hidden
* knowing which boards were hidden (split across multiple package directories...)

In contrast, keeping a list of favorite boards, and ensuring they are displayed at the top of the list, is a relatively small change.  Boards already have a unique ID (package + board ID) that can be used for the value to be stored in the favorites list.

Initially, just show the favorite boards at the top of the list, right?

**_Open question: How does a user indicate a board is a favorite?_**