arduino/Arduino #6035: Handle board change while compiling & uploading

PaulStoffregen, 2017-03-03:
The IDE allows users to change the board while compiling.

If the user clicked Upload, after the code is fully compiled for the old board, upload is started using the new board setting and an incompatible HEX file compiled for the old board.  For example, code compiled for Arduino Uno can be uploaded to Arduino Due.

This problem much easier to reproduce on slower Windows computers, where the compile process takes enough time for the user to change settings while the builder is still running.

PaulStoffregen, 2017-03-03:
Perhaps changing the setting could abort compiling?

Or maybe compiling could finish normally, but if settings have changed while compiling, simply don't upload?

per1234, 2017-03-03:
You can "brick" an ATmega32U4 based board by starting the upload with Uno, etc. selected in the Tools > Board menu, then changing to the correct board selection before the upload begins. I've actually done this. It was easy enough for me to fix but not so much for a beginner.

I think the best solution would be to upload to the board that was selected when the upload button was pressed, rather than the board that is selected at the time the upload starts.

Another option would be to disable the Tools > Board and custom tools menus until the upload completes.

PaulStoffregen, 2017-03-22:
Opps, sorry, errant clicking on the milestone info.

facchinm, 2017-03-22:
No prob :wink:

PaulStoffregen, 2017-03-22:
Looks like you're getting ready to release 1.8.2, right?

commit e6675739 (2018-05-14): Avoid board change during compilation/upload

By threading the boardChange callback we can busy wait until
the compilation/upload phase has ended and change the board when done.

Fixes #6035
commit 4c40302b (2018-05-15): Avoid board change during compilation/upload

By threading the boardChange callback we can busy wait until
the compilation/upload phase has ended and change the board when done.

Fixes #6035
commit 8270795f (2018-05-22): Avoid board change during compilation/upload

By threading the boardChange callback we can busy wait until
the compilation/upload phase has ended and change the board when done.

Fixes #6035
commit 8bbf92b8 (2018-05-30): Avoid board change during compilation/upload

By threading the boardChange callback we can busy wait until
the compilation/upload phase has ended and change the board when done.

Fixes #6035