arduino/Arduino #6036: variable not assignment value

Baytron, 2017-03-03:
unsigned long currentMillis = 0;
unsigned long prevMillis = 0;
int a = 0;
int i = 0;

void setup() {
  Serial.begin (9600);
}

void loop() {
  //Serial.print (i);  //sketch not work without Serial.print (i) or delay (5). variable "a" rise to max value 
  currentMillis = millis();
  if (prevMillis + 1000 >= currentMillis) {   //Cycles Count
    i = ++i;
    if (a < i) {
      a = i;
      Serial.println(a);
    }
  }
  else {
    i = 0;
    prevMillis = currentMillis;
  }
}

Baytron, 2017-03-03:
problem in IDE 1.6.5, 1.6.13, 1.8.1. With two boards UNO R3 (clone), NANO (clone).

facchinm, 2017-03-06:
Hi @Baytron , this is a question for the [forum](http://forum.arduino.cc/index.php) since your code needs a good refinement before calling a bug in the core :smile: If you post your problem there you'll get more guidance and advice than you can get here. Thanks!

closed 2017-03-06 by facchinm