arduino/Arduino #6037: I need help

radoslavov, 2017-03-04:
Simple test

double x;
double dx;

void setup() {
   Serial.begin(115200);
}

void loop() {
String fx("13.80");

dx = fx.toFloat();
x = 13.80;
Serial.print(x,6);
if(x == dx){
  Serial.print("True");

}
else{
    Serial.print("False");
  }
  }
  
The result is false - Why? When we transfer string to float and check with double if function does not work

mikaelpatel, 2017-03-04:
It is difficult to use == on floats due to the representation. The compiler might not use the same algorithm for the literal float in x = 13.80 as the Arduino String member toFloat(). Defined as double the compiler might truncate the literal value. In any case, simply check the difference between the values.

facchinm, 2017-03-06:
Hi @radoslavov 
if you also print `dx`, you'll get that its internal representation is slightly different from `x`, like `13.800000` vs `13.800001` . This is due to how floats are represented (see http://stackoverflow.com/a/12445693 for an overview).
However, this kind of question is very _forumish_, if you could post there a lot of people would participate in the discussion and the solution would be available for others instead than disappearing between hundreds of closed github issues :wink:

closed 2017-03-06 by facchinm