arduino/Arduino #6039: Broken eagle files for Arduino Micro

michaellass, 2017-03-04:
The eagle files linked on https://www.arduino.cc/en/Main/ArduinoBoardMicro are broken in several ways.

1. The line `<schematic xreflabel="Ð­^L^F" xrefpart="/%S.%C%R">` in Arduino_Micro_Rev03j.sch makes eagle output an error, unable to open the file.
2. After removing the broken attributes from this XML entry, cross referencing is broken as U2 comes from different libraries in board and schematic files.

A first-aid patch is attached but I'm not sure if these are the only issues in these files.
[Fix-issues-in-Arduino-Micro-Eagle-files.txt](https://github.com/arduino/Arduino/files/818831/Fix-issues-in-Arduino-Micro-Eagle-files.txt)

withoutian, 2017-04-04:
Turns out Arduino Nano's eagle file is also broken. Anyone know how to patch these two?

agdl, 2017-04-20:
Fixed both of them. Thank you

michaellass, 2017-04-22:
Seems like the fixed files haven't made it to the website, yet. Could you have a look again, please?

KateElizabeth112, 2017-05-25:
I'm having a problem with the Arduino Nano Eagle file that's currently on the website. Any chance you could let me know how you fixed it?

matthijskooijman, 2017-05-26:
@agdl, could you verify the fix?

alexanderturner, 2017-08-27:
Not working for me - seeing Nano files broken as of today.

alexanderturner, 2017-08-29:
Files are opening in Eagle 6 and Eagle 7 but seem to fail on Eagle 8.

CLECMan, 2017-09-15:
Element14 has same error. I'm using Eagle 8 too

Error:

line 74, column 25: Unexpected ''.

<schematic xreflabel="Ð­" xrefpart="">

srrobo92, 2017-09-27:
Any update on this? I'm trying to open the micro schematic in Eagle 8.3.2 and it comes up with the same error caused by that line.

winsonwq, 2017-10-28:
not sure if someone is working this ... any update?

agdl, 2017-10-30:
@winsonwq should be fixed. If so close the issue please

michaellass, 2017-10-30:
Well, the files for the Micro available under https://content.arduino.cc/assets/arduino-micro-reference.zip still show the same error as when I originally reported this issue. The link to download eagle files for the Nano does not work at all anymore: https://store.arduino.cc/arduino-nano.

Not my definition of a fix….

agdl, 2017-10-31:
The micro files works fine. Clear your cache. Same for nano

michaellass, 2017-10-31:
The dowload link for the nano files works again. Also, the cross referencing issue is indeed fixed. But still, Eagle 8 denies to open the schematic files with the following error:

    line 173, column 25: Unexpected ''.

Still, this is easilly fixable by removing the following attribute from the `<schematic>` element:

    xreflabel="Ð<U+00AD>^L^F"

Just to make sure, I have the most recent version, here are all checksums:

    e12a02e6e28192f9f7cf79e6e703da848e281996  arduino-micro-reference.zip
    64aa6236489a960e1e73b6559b3934750d18d974  arduino-nano-reference.zip
    b84ba966c83325ad6a5b41e00e0dedcecf01475c  Micro_Rev1j.brd
    3bf167a319992a0e0c65a602b340900d53dcb25a  Micro_Rev1j.sch
    9e9b1fb2f021315f2608eda3cff6f8f21334fbe1  NanoV3.2.brd
    9d3d10c990d7fa9dca0b90559a2a1558f8d47220  NanoV3.2.sch

agdl, 2017-10-31:
@michaellass we use eagle version 7 sorry for this. We are not planning to support eagle 8 at the moment

michaellass, 2017-10-31:
Alright, then I guess this can be closed. Thanks for fixing the cross-ref issue!

closed 2017-10-31 by michaellass

agdl, 2017-10-31:
@michaellass thank you for pointing this out!

ElectricRCAircraftGuy, 2017-12-27:
@michaellass where did you learn about the trick deleting the `xreflabel=` reference from the `<schematic>` element to fix the file? I just had to do this for a personal file made in v7.7 in order to open it up in v8.5.1, but I don't see this documented *anywhere* by Eagle.

alexanderturner, 2017-12-27:
Hey @michaellass, @ElectricRCAircraftGuy 

I had this exact same issue and it appears that it wasn't the `xreflabel` tag rather the incompatible character in `Ð`. I substituted this for a `D` which allows me to open the file in v8.5.1.

Strangé

Alex

ElectricRCAircraftGuy, 2018-02-03:
Nice find. I went back and deleted just that one bad character (shown above as the black diamond with a question mark in it) and now mine works too.

franxi2953, 2018-02-17:
https://drive.google.com/open?id=1BMtMBnjat-mZewCM2ewexx_O1TGkF09F
Fixed.