arduino/Arduino #6043: compiler warning setting wrong interpreted

trivalik, 2017-03-05:
If preferences.txt has the line compiler.warning_level=None
no warning is shown.
In case this line is missing, warnings will show up, but the Arduino IDE shows in preferences Dialog in combobox that no warnings should come.

facchinm, 2017-03-06:
`compiler.warning_level` is a compulsory field of `preferences.txt`. In fact, the IDE loads that file at startup and populates its internal state with the data. If the data is missing, it does unintended stuff (like the one you are experiencing) but it's totally normal since it's unable to properly assign a value to the selected dropdown (see https://github.com/arduino/Arduino/blob/master/app/src/cc/arduino/view/preferences/Preferences.java#L820) . I'd rater avoid patching a problem which arises when you manually remove compulsory preference lines :slightly_smiling_face:

closed 2017-03-06 by facchinm

trivalik, 2017-03-06:
I didn't remove it manually. I did it only to reproduce the issue. In fact the GUI should show the right if no entry is there.

facchinm, 2017-03-06:
But if you close the IDE and reopen it it restores the line and becomes self consistent, right? Did it get removed by itself the first time?

trivalik, 2017-03-09:
I used before Version 1.8.0 with another boardsettings ,which use with portable folder. But I have also a original one where I never touched this setting.
This Comes to my mind as I unzipped the 1.8.1 Version und applied the new board Settings ( just copy from 1.8.0). On my first build of Marlin, many warnings come up, then I opened the again the portable 1.8.0, and was still without these warnings. So I am not sure what trigger that issue exactly.

facchinm, 2017-03-09:
Well, this is (probably) a known bug of `arduino-builder`; if a build completes successfully the builder reuses the compiled files as a cache for the next compilations. Since it doesn't recompile the files, no warning is emitted on the second compile. This is not due to IDE's waning levels or similar, only a caching issue.