arduino/Arduino #6047: I2C - random crash - Nano

kb-, 2017-03-07:
I'm running I2C transfer tests on a multi master setup (2 Nano masters and one slave). [test code](https://github.com/kb-/Arduino-I2C-2way-anydata/tree/dd5597835c9a8a89ff23cc17605bb6f59faa5699)

The tests consist in transferring data both ways between the masters and the slave at an average 10 ms interval. I added collision handling to my code and checked the bus signal with a logic analyzer to be sure it's not overcrowded.
Data is transmitted fine with with a few (1 per 100k) retries when collisions occur, but **one of the masters ends up crashing after a random amount of transfers** (70 000 on average). I added timeouts to every loop in my code and in Wire.cpp to avoid endless loops but the crashes still occur, I'm thinking of a bug hidden deeper in the Arduino code.

I don't think the problem is material related because setting a different Arduino as slave leads to a different master ending up crashing.

I don't notice crashes with transfers between one master and one slave (an overnight test with 3 million successful transfers, same code with 6 ms interval)

wmarkow, 2018-04-09:
You can use [ArduinoCrashMonitor](https://github.com/Megunolink/ArduinoCrashMonitor) library to find the source code line which makes your Arduino to crash (hang). It helped me a lot when investigating I2C endless loops.

kb-, 2018-04-09:
Tricky I2C!
Thank you. These won't work for me unfortunately since the microchips of my boards are ARM based (SAMD21). I'll try them if I need to do some debugging on my Arduino 2560 tho.
I was looking into using the watchdog timer as well, but then decided to use OpenOCD with a Raspberry PI as a debugger.

I'm still having issues with this project. I started posting here since it's related to the SAMD library <https://github.com/arduino/ArduinoCore-samd/issues/222#issuecomment-378998498>

per1234, 2018-04-10:
Closing as duplicate of https://github.com/arduino/ArduinoCore-samd/issues/222. Please only create one issue per topic.

In the Arduino world when you say "Nano" we're going to assume you're talking about an Arduino Nano board, which is ATmega328P-based so in the future please refrain from the use of that term when referring to a board other than the Arduino Nano.

closed 2018-04-10 by per1234