arduino/Arduino #6050: HardwareSerial reset bug

gioblu, 2017-03-08:
CIao, I noticed a strange behaviour working with Serial. If a device restarts while other devices are transmitting data, can end up frozen:

TX:
```cpp
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("I will now freeze you!");
};
```

RX:
```cpp
void loop() {
  digitalWrite(13, HIGH); // Blink forever
  delay(10);
  digitalWrite(13, LOW); 
  delay(30);
};
```
in this code I have the receive making begin() and then blinking in the loop, if I reset the RX:
- often RX freezes with LED ON, seems before setup
- often RX freezes with LED OFF, seems before executing setup
- mostly I need to reset both at the same time to get back the receiver blinking
- oscilloscope shows TX data coming out so seems not freezed
- bug occurs with or without USB connected in RX or TX (so using other power supply)

Serial buffer overflow? Something connected to bootloader or because of serial incoming data in the process of boot after reset?
I have reported the bug in the italian forum: http://forum.arduino.cc/index.php?topic=460988.0

facchinm, 2017-03-08:
Hi @gioblu ,
it looks like a problem with optiboot indeed, since the Serial is not getting configured in the RX board. Would you mind trying updating optiboot via https://github.com/Optiboot/optiboot/tree/master/optiboot and report if the problem gets fixed?

gioblu, 2017-03-09:
Ciao @facchinm I do not have time to provide with tests, maybe this weekend

per1234, 2017-08-22:
@gioblu did you ever get the chance to try the latest optiboot? If so, did that solve the issue for you?

per1234, 2018-04-28:
@gioblu we're still waiting to hear about your results with the latest optiboot.

gioblu, 2018-05-21:
Ciao @per1234 I am sorry for not supporting the resolution of this issue for now. I hope to have some time in the next week to help you with testing.

hyndruide, 2018-06-25:
i try but it's doesn't work great i have to had a wait_RS485_pin_change(); to the end of startx()
and bus.strategy.set_RS485_delay(5); side arduino.

now i can send one byte but i can't send a string.

do you have a code  for rpi and arduino to try send a string  arduino ->RPI ?