arduino/Arduino #6057: Error acessin bool

plgarcia, 2017-03-11:
When setting bool to true or false, only the lower bit is set to 1 or 0.

But a test compares the full byte.

When for some reason, for example restoring à bool from EEPROM, some of the higher bits of the octet are set to 1, if (mybool) is always true even if the mybool is set to false jut before the test..
There is no way to set this bool to false.

For static variable this happens or not from one version to another, depending probably on what is present in memory at startup.

To solve that, in my software I declared some defines (BOOL uint8_t, TRUE 1, FALSE 0) and do not manipulate bool any longer.
I program on Arduino MEGA 2560

plgarcia, 2017-03-11:
Sorry for the error in the subject
accessing of course.

facchinm, 2017-03-13:
I believe this is due to EEPROM working at byte level , so storing a bool (which exists as a type in C++ and is 1 bit long) will only change the LSB of the storage byte. 
@Chris--A , what do you think about it?

cmaglie, 2017-03-13:
@plgarcia 
Can you show me a minimal sketch to reproduce the problem?

Actually `bool` *is* an `uint8`, maybe just a particular `uint8_t` that can theoretically have only two values (0 and 1), but it uses always one byte, so I'm wondering how you can actually store just 1 *bit* in the EEPROM.

As a proof this sketch:

```c++
void setup() {
  Serial.begin(9600);
  Serial.print("size of bool is ");
  Serial.println(sizeof(bool));
  Serial.print("size of uint8_t is ");
  Serial.println(sizeof(uint8_t));
  bool b;
  
  b = 255;
  Serial.print("after b=255, b is ");
  Serial.println(b, DEC);
  
  b = 0;
  Serial.print("after b=0, b is ");
  Serial.println(b, DEC);
  
  b = 254;
  Serial.print("after b=254, b is ");
  Serial.println(b, DEC);

  b = true;
  Serial.print("after b=true, b is ");
  Serial.println(b, DEC);

  b = false;
  Serial.print("after b=false, b is ");
  Serial.println(b, DEC);

  if (b == false) {
    Serial.println("conditional works");
  }
}

void loop() {
}
```

prints:

```
size of bool is 1
size of uint8_t is 1
after b=255, b is 1
after b=0, b is 0
after b=254, b is 1
after b=true, b is 1
after b=false, b is 0
conditional works
```

plgarcia, 2017-03-13:
Yes the first time I discovered this problem, it was when I restored a Boolean from eeprom. But I encountered the same problem with static data, bools in structs, .... 
I have printing values before an update, and it happens that whatever the update of the boolean I do (= true) or (= false), the upper bits are not affected. I understand that setting a Boolean in a field of Booleans needs to make bit operations.
It is up to the compiler to make behavior right.
mybool = false ;
if (mybool) {
	// a section of code that should not be executed
	// but unfortunately is
	Bla bla
}

I have spent too much time debugging strange behaviors due to that problem.
I solved my problem by not using bool, but that remains a bug.

plgarcia, 2017-03-13:
Yes and your test shows that even if you try to set 255 to the bool its value is 1 and upper bits ar not affected.
The problem is when in memory upper bits are not 0 they stay not 0 and the test of such a boolean is always true. (a boolean in the stack !
static bool mybool = false;
may not work! I have seen it!

cmaglie, 2017-03-13:
Let me state it more directly: if you can't provide a test program that shows the problem we can't help you.

matthijskooijman, 2017-03-14:
I think the cause here is that the compiler assumes that a bool has all bits zero, except for the LSB, which might be one. I believe this is standard C++ behaviour, so the compiler is "right" in doing this. In your case, you're probably loading a byte from EEPROM into a bool, and I suspect you are doing something like:
```
bool foo;
load_from_eeprom(some_address, (byte*)&foo);
```
e.g. casting the `bool*` to a `byte*` (possibly implicitly), which can cause the upper bits to become non-zero. I think the proper way to handle this is to load a byte from eeprom, and then convert that to bool (this should make the compiler properly convert the value: zero is false, any non-zero value becomes true).

I'll agree with @cmaglie here: Unless you can provide a small program that reproduces the problem, we can't really tell what's going on exactly (and my feeling says there is no bug here, but you're somehow violating the C++ standard and causing undefined behaviour).

plgarcia, 2017-03-14:
I tried yesterday to make a program that shows that with simple operations.
I have not been able, because the legal operations on boolean manage to wipe the higher bits, in all cases I have tested.
This problem cannot be trivial as it has not been identified earlier.
The two cases I have seen this are :
-	when reading data from eeprom, 
-	when using static bool variable.
The problem of the static appeared quite recently, the program growing uses 78% of memory and 69% of program memory, using areas of memory that may have been used before the program gets the hand.

I think the problem is that when affecting a bool (mybool = true) the only bit changed is the lower bit as it should update the full octet.
As the conditional operation test the full octet.

I will make some more tests to provide à very small program, but I will have to use EEPROM operations.

plgarcia, 2017-03-14:
The castings I have tried correct the data.

plgarcia, 2017-03-14:
I am not able to reproduce.
Can it be that some problems have been solved with the last version of the IDE?(compiler option ....)

per1234, 2017-07-01:
@plgarcia since you haven't been able to provide demonstration code for the issue and it's been some time I'm going to close this. Feel free to comment with a minimal sketch that can be used to reproduce the bug and I'll reopen.

closed 2017-07-01 by per1234