arduino/Arduino #6058: Error driver Arduino Mega 2560 Windows

plgarcia, 2017-03-11:
The driver delivered with the last versions of Arduino IDE does not work on windows for Arduino Mega 2560, but works with a Maga 2560 R3. I tested this on 2 different computers, Windows 8 and 7, and different boards 3 Mega 2560 not working I replaced by 3 Mega 2560 R3 working.
I do not envisage going back to an earlier version of the IDE as I need the new functionalities.

facchinm, 2017-03-13:
The Mega 2560 (not R3) uses the same driver of the R3. An official Mega 2560 exposes the following VID/PID combination: 
```
%mega2560.name%=DriverInstall, USB\VID_2341&PID_0010 
```
Anyway, once you install the drivers (also forcing the installation, if needed) you should be able to use your boards in any IDE version.

plgarcia, 2017-03-13:
Thank you for this information.
I reported this "bug" because of interest I think. My boards were working and are not any longuer since the update I made in August last year.
Up to the project what to do with this information.

Regards

SylvainLaurendeau, 2017-03-14:
Mega 2560 R3 drivers on WIN10 crash my PC and i can't anymore install on WIN10 ( new PC) .
MEGA2560 R3 driver not signed .
don't know what append it was working find for a long term on WIN10 . one day ,restart my PC  and get alarm saying drivers error PC close and try to diagnostic to repair , PC reboot itself and BANG! nothing append ,NO  BIOS loading ,PC still black screen . TRY all REBOOT control action and nothing , i have to send back to ASUS . Now using my old Laptop ( printer 3D with MEGA 2560 R3) and received same ERROR so i removed and try to find something worked. 
ANY option i can use to bring my 3D printer with ARDUINO  MEGA 2560 R3 !!!!!!

Best Regards

facchinm, 2017-03-15:
@SylvainLaurendeau I am 100% sure that our drivers are signed and won't provoke any damage to any PC.  Your problem is probably related with a virus or something, anyway it's not related with the IDE itself.

closed 2017-03-15 by facchinm

plgarcia, 2017-03-15:
I will make a test on mu daughter's Windows 10 PC neext week-end if it can help.
So I will be able to check if my problem persists on Windows 10 and to see if signature problem occurs.
Regardes

SylvainLaurendeau, 2017-03-15:
Can i attache the Mega 2650 R3 driver i have to be verify ?

Regards

SylvainLaurendeau, 2017-03-15:
See attached TXT files of my Mega 2560 R3 drivers 
Regards
[Arduino MEGA 2560 REV3.txt](https://github.com/arduino/Arduino/files/845169/Arduino.MEGA.2560.REV3.txt)

NetmaxIOT, 2017-11-14:
this is best way to  install arduino mega 2560 drivers
[(https://vtwood.org/how-to-install/how-to-install-arduino-mega-2560-drivers](https://vtwood.org/how-to-install/how-to-install-arduino-mega-2560-drivers)

kevykibbz, 2023-04-05:
Just bought my Arduino mega 2560 R3 but its not showing up as the selected default port on my windows 10 machine in Arduino IDE what should I do?