arduino/Arduino #6059: Please add the "RTCtime by smz" library

smz, 2017-03-11:
Hello,

I've just finished packaging my RTCtime library (https://github.com/smz/Arduino-RTCtime), which is meant to handle DS1307 and DS3231 RTC modules in the context of using the Standard C time library (AVR C Runtime 2.0.0, Arduino IDE 1.6.10 and later).

I'll appreciate if it could be added to the Library Manager.

Thanks in advance,

Sergio Manzi

P.S.: just let me know if there is anything wrong that must be fixed...

cmaglie, 2017-03-13:
Done!

closed 2017-03-13 by cmaglie

smz, 2017-03-13:
My first library, let's celebrate! 🎉  :confetti_ball:

Ciao e grazie!

Sergio

smz, 2017-03-14:
@cmaglie, sorry for hitting on this closed issue, but I made some n00b mistakes releasing my lib and I would like to check with you if they can be fixed, but at the same time I don't want to disturb the whole community with a new issue which is only due to my _noobiness_:

1. I named my lib "RTCtime by smz", without thinking (silly me...) that it would become the name of the folder containing the lib too. My ego isn't so inflated to wish my nickname to become part of the folder name, so, **can I modify the name of the lib to simply "RTClib"** in `library.properties`**?**

2. **Can I change the** `author` **and** `maintainer` **fields to just "smz"** (_once again, I'm not seeking publicity for my name..._)

3. If the answer to the above questions is "YES!", **should I tag and publish a new release** or can I just delete/untag the current release and then immediately re-tag a new 1.0.0 release?)

4. I had some "Alpha" releases which were flagged as pre-release and I thought they did not show up in the Library Manager, but... they did! I've now deleted/untagged those experimental revs. I hope this is OK.

Again, my apologies for disturbing and thanks in advance.

PaulStoffregen, 2017-03-14:
Just make your fixes, increment the rev in library.properties, and then tag a new release.  Easy peasy.

Do allow a day or two for the boards manager info to update, and check on another clean PC before sounding the alarm if it doesn't update.

smz, 2017-03-14:
tnx, @PaulStoffregen!

cmaglie, 2017-03-15:
Everything right except that a change in library name requires an admin intervention (in other words you can't change a library name once registered). BTW now I edited the name, so everything's fine :-).

For all the other fields instead you just need to change the library.properties and release again, as already explained.