arduino/Arduino #6062: Still issues with the hw serial library

mattskimi, 2017-03-12:
There is still serious issue with the irq driven version. 
Had to revert back to 1.0.6 where no issues.
The code is simple, collecting data from the analog inputs in an irq routine and sending those to a display through serial with 115200 baud. There is strict handshaking and something is dropped or transferred twice. Speed makes no difference, it crashes sooner with high speed.
Version 1.0.6 is the one  that works. Application  has been up and running for days. 
With 1.5 or newer the system crashes within 1 minute.

PaulStoffregen, 2017-03-12:
To have any chance to anyone ever looking into this, you must post a complete (hopefully compact) program which demonstrates the problem.  Of course, a likely response will be "don't do that from interrupts".  But without a program posted, nobody can do anything at all.

mattskimi, 2017-03-12:
Hi Paul,
I understand, and when I have time, I will try to create an app that
simulates this. There are propably a lot of coders suspecting their own
code working on workarounds.
/kimi

facchinm, 2017-03-13:
@mattskimi ,
a stripped sample code demonstrating the issue (plus an indication of the hardware/board you are using and the libraries version) is the minimum requirement to start analysing this, thanks :slightly_smiling_face:

PaulStoffregen, 2017-03-13:
While recently working with the Wire library, I noticed the Wire library "slave_receiver" example calls Serial.print from its receiveEvent() function, which the Wire lib calls from interrupt context.  This probably isn't a similar case, but it is one of the oldest and most well established of the many Arduino libs that has always demonstrated to users calling Serial.print() from interrupt context.

mattskimi, 2017-03-13:
Hi.
I only collect data from the adc  in the irq. Raw code and calculation. Not
a single library call.
/kimi

PaulStoffregen, 2017-03-13:
Well then, this only serves to show how we are spending our time with fruitless guesswork, because you haven't provided a program.

mattskimi, 2017-03-14:
This serves the users to try the older version of hwserial if they have
issues with irq driven code. The old works like a Buick. I will try to
create a simple sketch when i have time.
/kimi

per1234, 2017-08-04:
@mattskimi we're still waiting on that demonstration sketch from you...

per1234, 2017-08-22:
Closing due to lack of response from @mattskimi.

closed 2017-08-22 by per1234

mattskimi, 2017-08-23:
Hi, will reopen when i have the test sketch that proofs the issue.