arduino/Arduino #6071: ARDUINO 1.8.1 "IRsend" does not name a type

jedi345, 2017-03-14:
I try with the fix link for WINDOWS.zip but link do not exist any more??
I try to change board in settings....
I try to delete library IRrobot folder.....

facchinm, 2017-03-14:
Hi @jedi345 ,
we can't solve your problem if you don't provide:
- the pc you are using
- the sketch you would like to compile
- the board you are targeting

If the problem seems related with your configuration and it's not an IDE problem ,please write on the forum, thanks.

jedi345, 2017-03-14:
HELLO

-The PC ASUS  Windows 10
-Sketch IRsend demo
-Arduino Mega 2560 board (pin9)

When i Verify the sketch :exit status1
 "IRsend" does not name a type
???

TNX

facchinm, 2017-03-15:
The problem is related with https://github.com/z3t0/Arduino-IRremote#installation  (point 5). The library clashes with `RobotIRremote` in Arduino IDE so you need to remove it. 
Next time, enable Verbose compilation and ask the forum for assistance since this problem is not IDE-related, thanks!

closed 2017-03-15 by facchinm