arduino/Arduino #6078: can't start arduino ide 1.8.1

rpolasek, 2017-03-15:
no ide appears and no errors, it stucks with the messages below:
(os gentoo, 64bit, oracle jdk 1.8.0.121)

$ arduino
Loading configuration...
Initializing packages...
Preparing boards...
Starting...

facchinm, 2017-03-15:
Mmmh, this looks like one between:
- a connectivity problem 
- some configuration problem

In the second case, try deleting `~/Arduino` and `~/.arduino15/` folders.
If the IDE doesn't start either, try shutting down all your internet connectivity (disabling the interfaces for example) and report back the results (there could be some ongoing downloads in the main thread, locking everything up)

rpolasek, 2017-03-15:
i've already deleted ~/.arduino and ~./arduino15 and i've never have ~/Arduino. it didn't help. btw. i'm trying to execute arduino w/o any board connected to usb so my question is: is it required to have it connected before i execute arduino or not?

facchinm, 2017-03-15:
No, it's not needed. The fact that you are getting no output (so nothing is crashing indeed) is the strangest thing. Did you test without connectivity?

rpolasek, 2017-03-15:
i can't test it just now because i can't stop my connectivity. i'll try it tomorrow in the morning cet time.

matthijskooijman, 2017-03-15:
> The fact that you are getting no output (so nothing is crashing indeed) is the strangest thing.

I believe that this is a problem in the arduino-builder code that swallows up any exceptions during compilation. I already reported this to @cmaglie in a comment  somewhere, but I'm not sure if he fixed this already.

cmaglie, 2017-03-15:
> I already reported this to @cmaglie in a comment somewhere, but I'm not sure if he fixed this already.

I think that it's unrelated because that issue was about compile/upload from command line, instead here the IDE won't start at all.

rpolasek, 2017-03-16:
i don't think it helps you but i'm able to execute v1.6.8 but 1.8.1 always stuck.

matthijskooijman, 2017-03-16:
> I think that it's unrelated because that issue was about compile/upload from command line, instead here the IDE won't start at all.

Ah, indeed, so scratch my remark about that :-)

bgort, 2017-04-22:
After upgrading from a working 1.6.x Arduino IDE to 1.8.2, I'm having what may be a similar issue on Windows 8.1 x64.  IDE starts, but hangs at "Initializing Packages..."

Debug version is the same:
```
C:\Program Files (x86)\Arduino>arduino_debug.exe
DPI detection failed, fallback to 96 dpi
Loading configuration...
Initializing packages...
```

Sits like that forever.  There's no internet activity; using firewall to block the included java doesn't change anything.  I've deleted my Arduino15 directory - it was recreated (empty) on next launch, but still have the same issue.

bgort, 2017-08-03:
Not from here.  I ended up downgrading.  Didn't have time/energy to try to sort it out, unfortunately.

facchinm, 2017-08-03:
I can only think about: infinite loops (as in folders with circular symlinks) or infinite wait on some IO operation (like reading data from the serial ports) (very unlikely, almost all the code is wee guarded against these possibilities). @bgort does it also happen on IDE 1.8.0? We need some bisect from your side to track down the offending commit

bgort, 2017-08-03:
I'll try to try whatever the latest is, and 1.8.0, later today.

facchinm, 2017-08-03:
Thanks a lot

bgort, 2017-08-03:
1.8.3 works fine.  Started right up.  Go figure...

per1234, 2021-04-06:
Since the IDE didn't provide enough information to enable an investigation and later versions were reported to work, I'll go ahead and close this. Certainly the IDE should have provided more feedback, but I haven't noticed that specific issue with recent IDE versions (you might even argue that it now provides way too much information).

If anyone is still experiencing this problem, feel free to comment and I'll investigate.

closed 2021-04-06 by per1234