arduino/Arduino #6085: Option to hide console pane

antariandel, 2017-03-17:
It would be helpful to have an option to hide the console pane in the IDE (say with a button somewhere, or in the Tools menu).
This was already addressed in issue #2221 but was closed as Wontfix without any further explanation, so I'm reopening.

BrendaEM, 2017-03-18:
Thank you for re-opening it.

This would be a good improvement for debugging code as far as seeing structure--especially on laptops. It would aid in making demonstration videos by eliminating extraneous information.

I do not understand why the window could not just be dragged to bottom to at least reduce the console pane.

Changing the console lines in the user preference files does not seem to yield any change.

matthijskooijman, 2017-03-19:
> I do not understand why the user was ever prohibited window from dragging the console window to suit their needs/tastes.

Huh? You should be able to resize the console window by dragging the border, I do that often. You can't completely hide it this way, but you can make it smaller.

BrendaEM, 2017-03-19:
On 1.8.1 the user can't get any closer than than like 4 lines from the bottom of the screen.

Also, the the colored bar just above the window, which takes up an entire line, does nothing useful; the status bar is the colored bar at the bottom. The colored bar above the console should go.

matthijskooijman, 2017-03-19:
> On 1.8.1 the user can't get any closer than than like 4 lines from the bottom of the screen, on both Windows and Mac.

Right, that's true (so a button to completely hide it would indeed be useful). However, I was responding to your remark that the user was prohibited from dragging the console at all, which is something different.

> Also, the the colored bar between the editing and console window, which takes up more than an entire line, does nothing useful.

This line is used for error messages, and inputting things like filenames when renaming a file.

robzach, 2017-03-25:
I was also quite surprised to see this useful feature, which was in 1.6.4 and others, disappear. I make the editor font very large so I can teach Arduino to a class on a projector, and having ~10% of the vertical editing space occupied unnecessarily by a blank console is frustrating.

BrendaEM, 2017-03-25:
I've never seen the Arduino editing window go to the bottom.

BTW, just so you know, even a Arduino Micro holds more than 2198 lines. I would love to have those 4 editor lines, with the hope that I can cut my program down to 2197 lines.

agdl, 2017-03-27:
@BrendaEM if you want to pass from 2198 to 2197 you can delete one new line character. This should work ;)

natevw, 2017-06-05:
Yes, another use case would be on the PocketCHIP, where the Arduino IDE runs. It's not going to be an amazing experience even without the console, but would at least make it somewhat feasible for my son to use.

PR #6130 (2017-08-01): Allow setting low values as minimum console size
Fixes #6085 when `setting console.lines=0` in `preferences.txt`
commit 01902250 (2017-08-01): Allow setting low values as minimum console size

Fixes #6085 when setting console.lines=0 in preferences.txt
closed 2017-08-01 by facchinm